motioneye-project/motioneye #1704: FR: Possibility to turn cameras on and off

M0ebiu5, 2020-04-12:
I have multiple esp32-cam's - the cams have an mqtt interface which allows them to be turned on and off (en/disable the stream). One reason is to reduce the used bandwith on the wifi, the other is the power consumption (also on the server side).

Now it would be nice, if motioneye had the possibility to define on / off buttons for each cam, where the user could execute some sort of script / command, when pressing the button.
Another use case is, that an external motion sensor should be able to trigger the on/off/record command via an url.

jmichault, 2020-05-04:
to define buttons, look at https://github.com/ccrisan/motioneye/wiki/Action-Buttons

M0ebiu5, 2020-05-04:
Great, thanks for the hint!

closed 2020-05-04 by M0ebiu5