motioneye-project/motioneye #1706: Adding a camera creates problems

Fillip38, 2020-04-15:
Hello,

I am running on my Debiam 10 system motioneye 0.42. I only completely redesigned the system last weekend. I currently have 4 cameras running there. If I add a 5th now, then the overview of the camera shows me the image of the 5th camera in position 5 AND position three. The settings of camera 3 are correct, however, is called camera 3, it only shows the wrong picture, see screenshot ... Is there anything known?

[image]

[image]
[image]

ccrisan, 2020-04-15:
Your camera IP addresses and/or port numbers got mixed up somehow. Remove all of them and add them from scratch, one by one. If that' seems like a lot of work, verify your `motion.conf` and `camera-*.conf` files and check for streaming port duplicates.

Fillip38, 2020-04-15:
Ahh Perfekt. Yes Camera 3 and (the new) Camera 5 was the same Port 8085. Now i have change this ans it´s work!

closed 2020-04-15 by Fillip38