motioneye-project/motioneye #1712: Working schedule

recipriversexcluson, 2020-04-21:
Is there any method to run a working schedule so as to avoid certain times of day when running snapshots at intervals?

weemscreekstudios, 2020-10-19:
Similarly, I am interested in having the working schedule feature of motion detect also apply to the Movies option. I am using continuous recording in 5 minute intervals for daylight hours. It would be more automated to be able to schedule the time (like can be done now with Motion detection).

ananchev, 2021-01-17:
+1 for more flexible options for the working schedule. Currently even a relatively straightforward case to enable motion detection from e.g. 22:00 to 07:00 in the morning is not possible. Only you can define  00:00 to 07:00.

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #883, which asks for the working schedule to apply to interval snapshots; the same for continuous recording is #766. For a range over midnight, like 22:00 to 07:00: set the schedule to 07:00 - 22:00 and *Detect Motion* to *Outside Working Schedule*.

closed 2026-11-13 by JamBalaya56562