motioneye-project/motioneye #1713: timelapse files have the wrong extension

af7567, 2020-04-21:
Hi. I am running motioneye 0.42 on Raspbian 10. I can create timelapse files no problem and play them no problem but I was having an issue where Firefox would ignore my default mp4 application. 
It turns out the reason for that is that the MP4 timelapse file that is downloaded is actually an AVI file with an MP4 extension.
It seems that motioneye is setting the correct mime type so the browser knows it is an AVI, but creating the file with an MP4 extension.
I have tested it on Firefox and Chrome browsers and get the same result.

closed 2021-03-05 by starbasessd

af7567, 2021-03-05:
I'm assuming this was closed because there hasn't been any update on it for a long time, but the bug still exists. I have had a look into it and the problem is that the command to create a timelapse movie is using ffmpeg and the '-f avi' option so the file is always an avi.

mediafiles.py:695:
```
cmd += 'cat %(jpegs)s | ffmpeg -framerate %(framerate)s -f image2pipe -vcodec mjpeg -i - -vcodec %(codec)s ' \
               '-format %(format)s -b:v %(bitrate)s -qscale:v 0.1 -f avi %(tmp_filename)s'
```
But then the filename generated for the file uses the "Movie Format" set in the camera configuration, instead of always using avi

handlers.py:1269:
```
pretty_filename += '.' + mediafiles.FFMPEG_EXT_MAPPING.get(camera_config['movie_codec'], 'avi')
```
The Content-Type header is correctly set to tell the browser the data is video/x-msvideo though.

starbasessd, 2021-03-05:
What is version 0.42? motionEye is 0.41 for both motionEye and motionEyeOS
If you are running motion version 4.2, I'd recommend to upgrade to 4.2.1 or newer (4.3.2 is latest)

af7567, 2021-03-05:
motionEye was 0.42 when I opened the issue, it is now 0.42.1. I have just confirmed the issue still exists in 0.42.1.
I installed motioneye using pip

starbasessd, 2021-03-05:
Updated new issue to 'bug' and 'troubleshooting' to flag it.
Thanks for the updates.