motioneye-project/motioneye #1714: Cameras found, but no image

efyzz, 2020-04-21:
Hello,

I've installed Motioneye on my Zyxel NAS542 with Debian Jessie. After following the install instructions I can open the Motioneye webinterface and configure for cameras. If I put in the RTSP URL of a camera (same for another cam with MJPEG URL) the camera is found and can be chosen.

But I get no image from the cameras, only the "no camera" icon is shown. 

The URLs are working with VLC player, so they are correct.

I don't know where to start searching for the problem. Are there any error logs?

Thanks!

efyzz, 2020-04-22:
I found out how to get the log file ... here it is:

[motioneye.log](https://github.com/ccrisan/motioneye/files/4518852/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneye/files/4518854/motion.log)

zagrim, 2020-04-25:
Based on motioneye.log I'd say you have a recent ME version (0.42) with old version of Motion (3.2.12). Support for Motion 3.x was dropped in ME 0.40 so please try to upgrade Motion.

efyzz, 2020-04-25:
Hi,

Thanks for that hint! I tried to update motion to 4.x, but there's no build for debian 8 / armhf. Except for Raspberry Pi, but this one doesn't work. When I try to build it from sources I get a lot of dependency errors I cannot resolve ...

So I downgraded motioneye to 0.39.3. Now it says, the camera URL is not correct. But as said before, the URL is working with VLC player and the cam pops up automatically in motioneye's add camera menu when I put in the URL. So it should be correct.

My second (MJPEG) cam also shows no image, but there is no error related to it in the log.

Additional, I get ffmpeg errors. As described [here](https://github.com/ccrisan/motioneye/issues/490) there seems to be no chance without moving to motion 4.x :( But this is for recording only, right? 

Here is the log, perhaps someone has an idea why the RTSP URL is reported as incorrect and I still get no image from both cams.

[motioneye.log](https://github.com/ccrisan/motioneye/files/4532727/motioneye.log)

efyzz, 2020-04-25:
Solved!

After fighting the dependencies I could finally install motion 4 and everything works :)

But CPU load is always >180% with about 1 fps. So hardware seems to be too weak :(

Hint: It seems my MJPEG cam works all the time, but there was also a problem with chrome. I have to use Internet Explorer to see any image.

zagrim, 2020-04-26:
Some of the logs hint that you might benefit of having support for hardware encoding of h264 but if that's not available (both from the hardware and the ffmpeg build) you might be out of luck with the performance :(

As for the browser issue, did you upgrade MotionEye back to 0.4x? 0.41 has a fix for one browser related issue, but I think there have been some more also, you might find something by searching the issues here.

DunoCZ, 2020-05-25:
Stop motioneye, type ` rmmod uvcvideo` and after that `modprobe uvcvideo quirks=640` try changing 640 to something like 128 etc. (after changing start motioneye again and change value until you get good visual quality).
Credits: [https://stackoverflow.com/questions/25619309/how-do-i-enable-the-uvc-quirk-fix-bandwidth-quirk-in-linux-uvc-driver/25619508](url)

closed 2021-03-05 by starbasessd