motioneye-project/motioneye #1716: motioneye not saving photos or videos

turb0dog, 2020-04-22:
wondering if you could give me a hand, my motioneye is not taking snapshots or videos and it is also not saving them? tried to use different custom paths for media generated and have chmod ed the right directory still isnt working any ideas?

turb0dog, 2020-04-22:
i can however use the snapshot url to take photo and view it at 192.168.1.230:8765/picture/1/current

turb0dog, 2020-04-22:
using snapshot and curl but that only leaves the last picture taken doesnt store photos

HawtDogFlvrWtr, 2020-04-24:
anything in /var/log/motion.log or /var/log/syslog?

turb0dog, 2020-04-26:
yeah says failed to take snapshot for camera with id 1: not valid. but can’t find where that value is being set

turb0dog, 2020-04-26:
i also have two motion processes running at one time for some reason

turb0dog, 2020-04-26:
just saw in the syslog 

ffmpeg_set_outputfile: permission denied

where is this set at? motion or motioneye confit?

HawtDogFlvrWtr, 2020-04-26:
Where ever in the web interface you have the output location (File Storage menu) set to doesn't have permissions it seems. You can see the camera ID in /etc/motioneye/camera-1.conf. You can also see the camera ID from the web interface under "Video Device", though you can't change it there. You will have several instances of motion running. One for stream (back to the website), one for motion detection and then another when motion is triggered and needs to transcode the video.

Random8it, 2020-06-15:
Newly installed on Ubuntu 18.04 Server. Not saving files or snapshot images. Live shots in Web interface are very nice. Setup both the storage device - a raid on the server and then setup a FTP site on the same server. FTP test is successful. But neither location is indicating any movie stored. Have set up to create movie from motion detection and walked through several times. Q: would FTP test be successful if no write privilege? What user will need write privilege? I can add them.

zagrim, 2020-06-16:
@Random8it FTP test seems to be creating a new directory on the server, so the test should fail it if was about write privileges. You should probably create a separate issue for your problem, at least unless you are yourself sure it is about the same thing as turb0dog has (it's a good idea to check the log files for enlightening errors, anyway).

closed 2021-03-05 by starbasessd