motioneye-project/motioneye #1718: [QUESTION] Pass though video stream to web client without re-encoding?

AndreVallestero, 2020-04-23:
Does MotionEye do a simple pass-through of the cam video-stream if quality and framerate are maxed? What about if framerate is lowered? If motion optimization is on? Ideally, I would like MotionEye to not re-encode the originally ingested stream and rather pass it through when streaming it to the web client. Obviously this is not possible at lower quality settings since it would have to re-encode for a lower bitrate target, but I'm not sure if frame rate would also affect this since h264 bitstreams are reliant on keyframes and frame deltas which would change based on framerate.

My goal is to reduce CPU/VPU usage to a minimum by avoiding re-encoding since bandwidth is not an issue for me.

ilovelinux, 2020-04-29:
As far as I know, motion doesn't re-encode the camera stream if it supports MJPEG stream; if it doesn't, motion take a shoot for each frame and chain them together using MJPEG, It encode a video only when are recording (i.e. h264). Lower quality settings is just a scaling the frame and it should not be too much resource-intensive (it doesn't need re-encoding because, as i said, MJPEG is just a chain of JPEG images of the camera).
Changing the framerate, motion just a drop some frames during the stream. MJPEG is just a chain (if you are a developer you can figure it like a list, even if the encoding is more complex) and remove a frame doesn't need of re-encoding.
For more informations: https://en.wikipedia.org/wiki/Motion_JPEG

I'm not sure about the last part btw, maybe someone can give you a better answer.

AndreVallestero, 2020-04-29:
@ilovelinux My camera source streams are all in h264 over rtsp. Do you have any insight into whether or not motion re-encodes h264 source streams if allwed to output at max settings? It would be ideal if it could simply pass-through the video stream directly to web-clients.

ilovelinux, 2020-04-30:
Motion needs to extrapolate frame by frame from the stream in order to detect movement. If you don't care about that, have a look to https://github.com/ccrisan/motionpie/wiki/Fast-Network-Camera :wink:

ilovelinux, 2020-04-30:
As far as I know, this option is available only for Pi Camera streamings: https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera
I think the support will be extended later(?)

closed 2020-05-01 by AndreVallestero