motioneye-project/motioneye #1722: Cam DID number

WirelessMind, 2020-04-25:
Hello!
I have a camera (https://www.amazon.it/gp/product/B07VN4MSJH/) that **I use with YI iot app**.
I know DID number, ip and my YI account credentials, but **I don't have any cam credentails**.

Can I add this to MotionEye? How?

mmeah, 2020-05-10:
If it has an open protocol, then it should work. See if your camera can be connected to VLC or iSpy or any other program first. https://www.ispyconnect.com/search.aspx?q=Fredi is a great resource. From the quick googling I did, the camera url may look like this http://[IP]/videostream.asf?user=[USERNAME]&pwd=[PASSWORD]&resolution=[WIDTH]*[HEIGHT]

http:// | videostream.asf?user=[USERNAME]&pwd=[PASSWORD]&resolution=[WIDTH]x[HEIGHT]
-- | --

http:// | videostream.asf?user=[USERNAME]&pwd=[PASSWORD]&resolution=[WIDTH]x[HEIGHT]
-- | --

WirelessMind, 2020-05-11:
Unfortunately I only have the username and password of the account, not of the single camera. I guess it's not open...

mmeah, 2020-05-11:
If all you need is the IP address, you can use a tool like Fing to scan your network to find it.
https://play.google.com/store/apps/details?id=com.overlook.android.fing

WirelessMind, 2020-05-11:
I know the IP address.
The problem is that that camera doesn't have a username and password.
I can watch the video through its app and access it through the app's banknotes that are used for all the webcams in the account.
The webcam does not have "his" credentials.

closed 2021-03-05 by starbasessd