motioneye-project/motioneye #1731: Full screen view doesn't work all the times anymore

JBS5, 2020-05-01:
I did not have any problems viewing a camera in full screen, but using this feature in the last couple of days, it is not working as expected anymore.

When toggling full screen mode, the camera is shown as below. After some refreshing and/or resizing the browser window, the full screen view pops up. 

[image]

[image]

[image]

Using Firefox 75.0 (64-bits) and Chrome 81.0.4044.129 (Officiële build) (64-bits)
motionEye 0.41 on Ubuntu 18.04.

Does anyone recognize this?

twinstef, 2020-05-25:
I have the same problem.

JBS5, 2020-06-06:
Not having the problem anymore, using Chrome now 83.0.4103.97 (Officiële build) (64-bits) and Firefox 77.0.1 (64-bits) now. Maybe it was browser version related?

closed 2021-03-05 by starbasessd