motioneye-project/motioneye #1738: rpi multicam with master hub

jarmodb, 2020-05-07:
Hi guys,
I’m new tot he raspberry pi and eyemotion stuff, but I want to create something. I want to create a multi security camera setup with 3 cams running on a rpi zero w each, connected to a master hub, an rpi 4. I don’t want the hub to save the video recordings to an HDD, but i only want it to upload the files to dropbox,  without the rpi zero’s uploading to dropbox itsself? What’s the best way to achive that? is it even possible?
Thank you, and stay safe!

CRImier, 2020-05-07:
This should work. Flash your Pi Zeros with MotionEyeOS SD cards and configure them as network cameras that don't do motion detection by themselves, then flash your Pi4 either with MotionEyeOS or with Raspbian and MotionEye, connect camera streams from Zeros to MotionEye on the Pi4, then enable motion detection on Pi4 and use Dropbox as the place to save recordings. I don't recall if MotionEye has Dropbox integration, but if it doesn't, shouldn't be that hard to make it work by installing some kind of Dropbox client on Raspbian.

jarmodb, 2020-05-08:
thanks, that worked. I've set up my zero cam as an fast network cam and in my hub i added it as an network cam. Now i can upload it to dropbox.

thank you!

closed 2021-03-05 by starbasessd