motioneye-project/motioneye #1747: Delay in RTSP feed (40-60 seconds)

Christiandk1986, 2020-05-12:
Hi

First of all, great software.

I am using a Reolink 410 with an RTSP feed. The feed runs with only slight delay (2-3 seconds) in VLC, but in MotionEye the delay is about 40-60 seconds on the same feed. What can be done?

hajosch08, 2020-05-13:
Hi
Have the same problem with INSTAR 5905HD. Tried everything with NTP helped nothing.
Need help too.

ozett, 2020-07-21:
yeah have also a rtsp latency.... if i grab the stream with gstreamer on the same maschine (jetson nano) it is ok, on the gstreamer pipeline i have also an option to choose udp/tcp for rtsp and choose a latency value... all missing in motioneye config for rtsp

jasaw, 2020-07-22:
On MotionEye, UDP/TCP option for RTSP is only available when you add a new camera. To change the UDP/TCP after a camera is already added, you'll have to edit `netcam_use_tcp` option in `camera-n.conf`. Replace the `n` with your camera number, like 1 or 2 or 3 ...

ozett, 2020-07-22:
great, thx, i will try...

cesarandreslopez, 2020-09-24:
@ozett @jasaw did you get better results with 
netcam_use_tcp False
or 
netcam_use_tcp True
for rtsp?

cesarandreslopez, 2020-09-24:
From my side, in case this is useful.

With MotionEye, I've found that you should use mjpegs instead of rtsp wherever possible if you need low latency.

RTMP seems fine always.

jasaw, 2020-09-25:
@cesarandreslopez I have to set `netcam_use_tcp on` because UDP was dropping too many packets, causing the video to break and pause. TCP seems to work fine and latency varies from 1 second to 5 seconds, which is not too bad.

Pedder007, 2020-11-14:
Hi I'm also using 2 Reolinks RLC-410Ws. Having it now for roughly a year running on a Raspi3B, and having yet never found a setup with a kind of fluent videostream and without delay, neither with http, nor with rtsp/rtmp, I decided to move to a Raspi4B (much better GPU + NW). Delay of mostly something around 3-6 seconds, partly up to 10, was also always an issue, as well as fragmentation by using rtsp/rtmp.
Moved now two days ago to the Rasp4B (completly fresh installation on basis of what is given here in the repository, all problems have gone form one day to another :-)
Configuration Cams via ReoLink-SW:
- StreamingTyp: clear
- Resolution: 720p
- Frames: 16fps
- Data: 2048 max kbps
- H.264: Basis
Configuration MotionEye:
- RTMP Stream (type: network cam, port: 1935, user+pw direct in the link)
- Resolution: 1024×576
- Frames: 10fps
- Streaming: 8fps
- Quality 75%
- Image Resizing: On
- Resolution Quality: 75%
With 2 cams, incl. streaming and motion detection, the Raspi4B now runs on ~ 35-50%.
No delay, no fragmentation and fluent streams, at least seems now only depending on my other network traffic, resp. on the poor network range of the cams. For the last aspect I really re-arrange my WLAN router setup a while ago.

Hope that might help for others :-)

closed 2021-03-05 by starbasessd