motioneye-project/motioneye #1751: Motion Detection video length question

sts-ryan-holton, 2020-05-16:
Hi, I've got a fast network camera set up recording into a centralised hub. I've got motion detection turned on and have a question regarding video length.

I've got a mask set up, which is probably covering around 40% of my video, I've got a resolution of 1024x576.

I'm trying to get the camera to **detect small amounts of motion, and then continue recording for X amount of seconds, say 20 or 30 seconds.**

How would I achieve something like this?

My current settings are:

[image]

zagrim, 2020-05-17:
Hi,
Do you want to record a specific length regardless whether there's still motion being detected or not? In that case you should probably have "Captured After" equal or greater than the amount of frames captured to get you enough footage so that the time set in "Maximum Movie Length" will be met first. Say, you get 4 frames per sec and want to have at least 20 sec of footage even with the shortest possible detected motion event, then you'd go for 20 sec in "Maximum Movie Length" and something like 4 x 20 = 80 frames for "Captured After".

nicemachines, 2020-05-18:
Thanks Zagrim!  Here I thought that "Captured After" meant the amount of frames to capture after there were no motion.  Geez, I am stupid!  Great program here if only some things were fixed, like the e-mail feature, that still doesn't work and I have tried everything.  Even the gmail.  DOESN'T WORK!

sts-ryan-holton, 2020-05-27:
Is anyone able to simplify this please?

zagrim, 2020-05-27:
Was my answer too complex (or non-understandable), do you mean the way capture length is specified (mixed use of "frames" and "seconds", and/or the amount of options related to capture length, or something else?

closed 2021-03-05 by starbasessd