motioneye-project/motioneye #1755: MotionEye does not show image but records ok

Ale029, 2020-05-16:
Hi,
I´m running Motioneye last version in Raspian buster, RPI 4b. Motion is working fine, records on motion, but Motioneye doesn´t show the camera image. In the log I see Serve returned 401 Unauthorized. How can it be if it´s recording? 
I´ll appreciate any help.
Thank you
[motioneye.log](https://github.com/ccrisan/motioneye/files/4639412/motioneye.log)

nicemachines, 2020-05-18:
Not sure Dude!  I have been trying to figure this program out and a lot still don't work right.  Hopefully he hasn't given up on this MotionEye because it is a great start!  I worry because nothing has happened since December of last year (2019).  Don't give up, press on!

KeithMacDonald, 2020-06-01:
@Ale029 - thanks for raising this issue. 

Seems like I have the same problem here. Motion detection is working fine, and capturing images. motionEye loads in the browser, no problem, but no image is visible in motionEye. Even so, when I click "take a snapshot", an image is created in the same folder as the motion detection images.

Anyone else seeing this issue? Anyone found a solution?

motionEye Version 	0.39.3
Motion Version 	4.1.1
OS Version 	motionEyeOS dev20181126
Using a Pi Camera.

closed 2021-03-05 by starbasessd