motioneye-project/motioneye #1756: Difference between simple MJPEG and Network camera?

Bootscreen, 2020-05-17:
Hello,

My current Setup:
FREENAS Server with a Motioneye Jail, the central. 
2x Raspberry Pi Zero W with MotioneyeOS
1x Raspberry PI 1 B with MotioneyeOS
1x PC with a Webcam as MJPEG Stream

The PI's are configured as fast network cams. 
And in the central they are configured as network cams.
Today I add the PC Webcam and See the simple mjpeg type. What is the difference between network and simple mjpeg? Espacially what need lesser Ressource or has the better Performance? 
I dont need motion detection, movie or Screenshot record or streaming. Only the overlay is nice to have. 

Greetings
Bootscreen

joshhodgson, 2020-06-04:
"Network cameras (or IP cameras) are devices that natively stream RTSP or MJPEG videos or plain JPEG images. Consult your device's manual to find out the correct RTSP, MJPEG or JPEG URL."

"Adding your device as a simple MJPEG camera instead of as a network camera will improve the framerate, but no motion detection, picture capturing or movie recording will be available for it. The camera must be accessible to both your server and your browser. This type of camera is not compatible with Internet Explorer."

MJPEG doesn't do any processing on the server and the browser is what goes to get the camera feed, so performance will be better

thisisbenwoo, 2020-09-16:
Follow Up question - I'm trying to access my Raspberry Pi camera at http://<ip address>/webcam/?action=stream. If I set this URL as a Simple MJPEG camera, it works, but if I define it as a Network camera, it shows no picture. Can someone help as to why?

joshhodgson, 2020-09-16:
Probably because you can access the image URLs from your browser but the MotionEye install can't access that URL itself

thisisbenwoo, 2020-09-16:
Huh, so for that particular camera which is a stock standard RPI-CAM-V2, how can Motioneye pull that stream? All I really want to do is to be able to flip it, as the camera is mounted upside down.

joshhodgson, 2020-09-16:
My guess is there's an issue with using the hostname. Does doing with IP work?

thisisbenwoo, 2020-09-16:
Same problem with IP address

*Ben Woo*
(212) 724-1188
*this*is*ben*woo*@gmail.com <http://gmail.com/>*

closed 2021-03-05 by starbasessd