motioneye-project/motioneye #1757: Don't save to SD card

sts-ryan-holton, 2020-05-17:
Hi, I'm uploading my footage to a local NAS server, I need to turn off saving to the SD card as I simply do not have enough storage space on the card to save locally, I just need to save footage to a NAS server, my current config which saves to my NAS server is attached in a screenshot.

If someone is able to tell me how to turn off saving to a local SD card that would be great, thanks.

[image]

nicemachines, 2020-05-18:
I have been messing around with this program with a few Raspberry Pi Zero W's in a Vilros kit (camera mounted inside zero case).  I got this program to save all pictures and video on an external 32 GB Samsung FIT.  I have had no luck in getting this program to e-mail me.  A simple task but no go.  Hopefully he is reading these posts and is still working on this MotionEyeOS.  Doesn't seem positive, no new versions since December 2019.  Only way to save your stuff and NOT use the micro SD is to hook up an external storage and tell the program to use the entire storage.  Try what I did.  The Samsung FIT are great USB storage devices.  The one I have is USB 3, even though the zero only has usb2.  I guess keep trying!

zagrim, 2020-05-20:
@sts-ryan-holton Just as nicemachines said it, the file storage you specify in ME will be used to save the video and snapshots. Usually the throughput to any network storage would be limited and less stable as local storage (external or internal), and thus it is a good idea to first store stuff locally before uploading to somewhere else. It also avoids immediate data loss and/or service halt in cases where network connection fails temporarily. And finally, the Motion daemon, which is the process actually processing and saving the video, doesn't have support for network storage.

What MotionEye doesn't have, I think, out-of-the-box is a clean-up for the local storage after uploading, but you could probably solve this by using a shell script to clean up files older than some safe interval to allow uploading to happen before the clean-up.
The script could be run outside of ME scheduled via `cron`, or by having it set up in "Motion Notifications" -> "Run An End Command" (so that it runs after every motion event).

zagrim, 2020-05-20:
@nicemachines I don't see anywhere a ticket from you for the email issue you are referring to. Please create an issue for it and stop talking about it in other issues. And yes, MotionEye is being developed, but as it is nobody's day job, you can't expect any steady paced progress.

zagrim, 2020-05-22:
I realised I goofed up above @sts-ryan-holton: No need to fiddle with a script to keep local storage cleaned up, it should be enough to set both "Preserve Movies" and "Preserve Pictures" to "For One Day", and you shouldn't have issues with filling up your SD card (unless it's very small in capacity).

cinnamonsweets, 2020-05-30:
I enable storage device network share through the config file so it saves to my NAS and not my microsd.

`sudo nano /etc/motioneye/motioneye.conf`
 media_path /dev/null
 log_level quiet
 mount_check_interval 30
 smb_shares true
 enable_reboot true

Then I try to minimize microsd wear by modifying my fstab to send temporary data and log files to ram instead of microsd card. You may need to trial and error the size for /tmp because lack of space can cause errors.

`sudo nano /etc/fstab`
 proc            /proc              proc   defaults,noatime                                  0  0
 /dev/mmcblk0p1  /boot              vfat   defaults,noatime                                  0  2
 /dev/mmcblk0p2  /                  ext4   defaults,noatime                                  0  1
 tmpfs           /tmp               tmpfs  defaults,noatime,nosuid,mode=0755,size=50m        0  0
 tmpfs           /var/tmp           tmpfs  defaults,noatime,nosuid,mode=0755,size=1m         0  0
 tmpfs           /var/log           tmpfs  defaults,noatime,nosuid,mode=0755,size=1m         0  0
 tmpfs           /var/spool/mqueue  tmpfs  defaults,noatime,nosuid,mode=0700,gid=12,size=1m  0  0
`sudo dphys-swapfile swapoff`
`sudo systemctl disable dphys-swapfile`
`sudo reboot`

closed 2021-03-05 by starbasessd