motioneye-project/motioneye #1758: Wrong color

RPiDespair, 2020-05-17:
Hi,

**Configuration**
Two RPi 3b
No. 1 RPi 3b with NoiR V2 camera
No. 2 RPi 3b with V2 camera

SD card from No. 2 Rpi 3b.
motionEye version 0.42
Motion version 4.2.2
OS version Raspbian 10
Running As Normal User

SD card from No. 2 RPi 3b is now in RPi No 1.

The camera image is displayed in pink.
The RPi No. 1 is on a tree and can only be reached via wi-fi.

A problem solving would be great, thanks.

CRImier, 2020-05-17:
Do you have a photo? It might be because the NoIR camera lacks the IR filter, which does change the colors quite a bit towards red, in my experience.

RPiDespair, 2020-05-17:
Before changing the SD card, the camera image was black and white (normal for the NoIR).
[image]

CRImier, 2020-05-17:
Hmm, AFAIK the normal image for the NoIR camera is in color - unless you change the settings to B&W, I suppose (if that's an option). Any chance you hotplugged the camera - plugged&unplugged the camera while it was still running? Changed some settings to make the image black&white on the original No. 2 SD card? Do you have an IR light lighting up the bird? This looks like what a picture would look like with an IR light illuminating the bird.

RPiDespair, 2020-05-17:
No. 2 previously worked optimally with the SD card and the V2 camera.
With no. 2 no adjustments were made to influence the color.
No. 1 with the SD card from No. 2, NoIR V2 camera and infrared light creates pink camera images. I only changed the SD card.

I apologize if the translation from German into English uses the wrong grammar.

cap9qd, 2020-05-18:
Every rpi noir camera I have setup in motioneye for the last 3 years over
several versions looks like the image you posted. I changed the motion
options to iridescenes (I think) to improve but all my noir look as you
posted. The noir sensor is the same as far as I know but just doesn't have
the ir filter in the lens.

RPiDespair, 2020-05-18:
Hi cap9qd,
No. 1 has worked with
[image]
and the NoIR camera very well since 2018. A pink photo was occasionally taken in daylight. Daylight is a problem with the NoIR camera. After changing the SD card, only pink photos are created. Even without daylight.

cap9qd, 2020-05-23:
It is interesting my father has the same issue after updating his csmera. Doing some more looking around it may be the AWB algorithm changing in the latest firmware.

https://github.com/raspberrypi/firmware/issues/1167

I haven't a clue what to do about it in motioneye or motioneyeos but this may point you or someone else in the right direction.

cap9qd, 2020-05-23:
I changed my NOIR cameras to exposure mode Grayworld and they are back to normal. I had an old version of motioneyeos on one ZeroW camera that I updated to have the grayworld option.

RPiDespair, 2020-05-24:
Hi cap9qd,
thank you for the interesting link. However, I lack the necessary knowledge to apply the workaround. The RPi No. 1 is on a tree and can only be reached via wi-fi. If I make a mistake, I can only get the SD card out of the bird's nest in a month.

cap9qd, 2020-05-24:
You should be able to change it via the web interface.

RPiDespair, 2020-05-24:
Hi cap9qd,
thank you for the screenshot. I cannot find these settings in my motionEye installation.
[image]
I probably need newer versions?

cap9qd, 2020-05-24:
Ah I have mine set for a fast motion camera so I have extra picture
options. I found a way to set the AWB mode via the extra motion options. I
don't do this do I can't test right now.

Found this:
https://www.gitmemory.com/issue/ccrisan/motioneyeos/2167/546670962

RPiDespair, 2020-05-24:
Thank you for the suggestion, but does that also work with motion version 4.2.2?
If I understand the conversation [https://github.com/raspberrypi/firmware/issues/1167](url) correctly, then I have to install motion 4.3.0 so that I can use the greyworld option. I have the instructions  [https://motion-project.github.io/motion_build.html](url)
looked at, but I don't dare to do that.

cap9qd, 2020-05-24:
Ah I'm sorry. I guess I assumed since it was in the fast motion you'd have
the option on motion. Not sure but someone else may have more experience
with it.

Id I have some time I'll try it on my spare ZeroW.

Pi-DD, 2020-05-25:
@RPiDespair - I’ve had to upgrade a pi in a remote location where I cannot access it to motion 4.3 and it worked just fine. Did the same for a Pi 0 running it in a nest box and that also worked. 
No guarantees, anything might be different in your setup, but I don’t think the risk is high!

RPiDespair, 2020-05-25:
The NoIR camera usually takes photos in the nest box when there is some daylight with a low color content.
Are your photos normal after the change to motion 4.3.1 or are they only black and white?

I successfully installed Motion and motionEye a month ago after the wiki, but I don't know enough about Linux.
How do I do the two commands
`wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb`
`dpkg -i pi_buster_motion_4.2.2-1_armhf.deb`
change that motion 4.3.1 is installed?

Pi-DD, 2020-05-25:
The NOIR camera will (in natural light or artificial light) give colour images- but the colours will not be “true” because it is still emphasising the red end of the spectrum. 
I have ordinary LEDs as well as IR ones in the box. 

I’ll post an additional comment about the commands when I can use a full keyboard!  (This from my phone. )

RPiDespair, 2020-05-25:
Can it be that I only have to adjust the path and the download?
`https://github.com/Motion-Project/motion/releases/download/release-4.3.1/pi_buster_motion_4.3.1-1_armhf.deb`
`dpkg -i pi_buster_motion_4.3.1-1_armhf.deb`
Is it necessary to uninstall motion 4.2.2 or is it enough to end motion and motionEye?
What is right?
I installed Motion and motionEye a month ago according to these instructions.
[https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian](url) and then
[https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User](url) so that I could delete the many photos.
If motion is updated, the settings made in motionEye will probably be lost?

Pi-DD, 2020-05-25:
You are correct, you need to alter the path as you have done and run the commands having put “sudo” before them as per the instructions you already followed. You have no need to uninstall the previous version. 
Updating Motion makes no difference to the settings for Motion👁  or the pictures already stored.

RPiDespair, 2020-05-25:
Switch off motion.
`sudo service motion stop`
Switch off motionEye.
`sudo meyectl stopserver -c /home/youruser/motioneye/motioneye.conf` (Running-As-Normal-User)
Is that correct?

Pi-DD, 2020-05-25:
You can use
sudo systemctl stop motioneye

The after update
Sudo systemctl restart motioneye

Similarly for motion

RPiDespair, 2020-05-25:
[image]
Thank you very much,
with your help, the installation of motion 4.3.1 was successful for my no. 2 RPi 3b with V2 camera.
How do I get "Automatic White Balance" added with the Greyworld option?
`mmalcam_control_params -awb`
Is that correct?

Pi-DD, 2020-05-26:
You need to put 
mmalcam_control_params -awb greyworld

Click on Video Device in the web control interface. Then Into the box in the web interface labelled “Extra motion options” that is revealed type the above and then click in the orange APPLY button that appears at the top of the interface screen.

RPiDespair, 2020-05-26:
Hi PI-DD,
it's done, thank you very much.
It is strange that the option appears in your photo in two lines. I use Firefox.
[image]

Pi-DD, 2020-05-26:
That screen shot is from Safari on my iPad. 

It’s confusing that different browsers render things in different ways.

RPiDespair, 2020-05-26:
Does the V4L2 camera have more options in motionEye? I do not know if there were any other options before setting the “Extra motion options”, which I had previously integrated my NoIR camera as V4L2.
[image]
Please refer
[https://github.com/ccrisan/motioneye/issues/1758#issuecomment-633257606](url)

That is with your "subtenants" is very unfortunate.
In total I have 6 nesting boxes for starlings. All nesting boxes were occupied and the first young birds flew out. After cleaning the boxes, all are occupied again.

Pi-DD, 2020-05-26:
I’m afraid I don’t  know about that! I’ve never set up one of my cameras in that way! The link in your post did not work for me. 

It may be that in that case Motion options have to be set directly in Motion itself? (Because, as I understand it, Motion is what passes parameters to the Pi camera software raspivid command.)

I might explore the alternative setup on one of my spare Pi’s, but unfortunately don’t have a spare NOIR camera. However, there are many other AWB and image settings that one could experiment with. 

https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

RPiDespair, 2020-05-26:
I don't know why my links don't work. Simply copy and transfer it to the browser.
See above from cap9qd commented 2 days ago.

RPiDespair, 2020-05-28:
MotionEye is currently showing a still image in the FF that is already 35 hours old. What could be the cause?

closed 2021-03-05 by starbasessd