motioneye-project/motioneye #1764: Higher resolution for Pi HQ Camera

azuma82, 2020-05-24:
Hello,

I like motioneye, I used for some bird pictures.
I want to use the maximum resolution, but, I think the stream is 1 reason, they’re are not working.

I have a picture for example, more resolution will be perfect.

Thanks  to the past work on motioneye! 
[image]
[image]
[image]

zagrim, 2020-05-25:
Nice pics, but what's exactly your question? What do you mean with "I think the stream is 1 reason, they’re are not working"?

azuma82, 2020-05-25:
Hello,

The question is, how I can use a higher resolution? Moitioneye stop working when I go higher then 1920 x xxxx. The stream is not my priority, so I don’t need it, but if it possible I want a higher resolution for the picture.

zagrim, 2020-05-25:
Does "stops working" mean "doesn't respond" or "video stream stops showing", or are there some actual error messages? My first guess is that if you are running MotionEye on RPi, doing motion detection on high-res (and high fps, I guess from your usage scenario) video just saturates the CPU.
Have you tested if it works any better when no browser session is active with MotionEye dashboard (meaning there's no application pulling the video stream)? Motion detection needs to decode the video, there's no way to avoid that, but if nothing is requesting for the picture stream for the camera, there's at least some decrease in the encoding side.

BTW, are you saving video or snapshots when motion is detected?

I don't know what codec the Pi HQ camera uses and whether that could possibly be decoded (and/or encoded for stored material) using specific hardware instead of doing that with software in CPU. With Raspberry Pi, the options are anyway quite limited.

If everything else fails, you could set up the RPi with the camera to act as a "remote camera" for another MotionEye instance (which is running on a more powerful hardware, with possibly better options for using hardware acceleration for decoding/encoding) where you actually do the motion detection and storing of the resulting files.

azuma82, 2020-05-25:
Thanks for you're support, I will test this. To you're questions, I only save the pictures. The error from Motioneyeos that I got was nothing, just no picture. But now I understand, it doesn't difference between Stream and Picture. So I must find I solution for that. 

There is no way tho stream in low resultion an doing pictures in high? 

running with a raspberry pi 3a+ the hq camera

zagrim, 2020-05-26:
For network cameras that are able to produce streams in several resolutions, one can use `netcam_highres` and `movie_passthrough` (which would also seem to mean "picture passthrough" in case of saving snapshots according to [Motion docs](https://motion-project.github.io/motion_config.html#picture_output)) to save the hi-res version without en/decoding it while doing motion detection with the low-res version, but I assume that can not be used with the Pi camera, since it is not a network camera. But then again, I've got no experience at all with anything else than RTSP network cams myself, and I might not know some of the tricks.

zagrim, 2020-05-26:
Since you say you're using Motioneyeos, this should help if you decide to try the "remote camera" approach: https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera

closed 2021-03-05 by starbasessd