motioneye-project/motioneye #1766: Amount of Action Buttons

BurtMcCloud, 2020-05-25:
Hello,

I've made some action buttons for my camera setup and everything is working swell except for one thing: the number of action buttons seems to limited to 8. I've made more and all of them won't show, only the "first" eight.

So it's not a bug, it's not anything causing error, it's a feature that I hope would do more: Allow to create more than 8 action buttons.

(ok, maybe I have missed something and it's actually my fault. Guide me please.)

Thank you in advance!

weiserhei, 2020-05-26:
This has been discussed in the past, for example see: https://github.com/ccrisan/motioneye/issues/844
I would also like to add some more 😁

davebdb, 2020-05-27:
One thought of a workaround is have a webserver also run on the system and have additional action buttons placed there to execute the commands you want to run.  Since I don't have a VPN yet, I have even SSH'd into the box through my complex access method and issued the commands via CLI, but if you want to use webaccess only, running a webserver on another port is probably your best bet of a workaround.

closed 2021-03-05 by starbasessd

MichaIng, 2024-05-28:
Fixed with: #2903