motioneye-project/motioneye #1782: Question: Using multiple NIC on MotionEye

austwhite, 2020-06-04:
Apologies if I have missed this somewhere, but some searching has not yielded results for me.
Basically, I want my cameras completely separated from my LAN.
The setup:
Computer with two Network Interfaces running MotionEye.
Separate 8 port PoE switch for the cameras, connected to one of the NICs
Second NIC to provide access to MotionEye interface and streams generated by MotionEye to the main LAN.
Cameras should not be able to see the main LAN and should remain isolated and only see MotionEye.

Is this possible?   If there is documentation, can you point me to it?

CRImier, 2020-06-04:
In fact, this setup is functionally the same as "motioneye server is accessible through VPN". Make sure there's a DHCP server on the internal LAN, that's about it. Can also firewall the interfaces, but it isn't that important.

CRImier, 2020-06-04:
Which is to say, yes, it is possible and simple enough to not require ME-specific documentation.

austwhite, 2020-06-04:
I'll keep looking with additional search criteria

CRImier, 2020-06-04:
I'd suggest you just start building it, no need to look into it that much, except for hardware choices, perhaps.

austwhite, 2020-06-04:
I have the hardware already.  I am just struggling with how to setup MotionEye so the cameras are on a separate LAN.
A lot of surveilance systems isolate the cameras from the main LAN.  Buying the premade ones, the software provides the stream to the LAN, the cameras do not connect directly to the LAN.
To implement that in MotionEye would require one network interface to connect to my PoE switch for the cameras, on a completely different subnet to the interface that connects to my LAN.
I can set it up single NIC, no drama and that is really simple.    It is trying to work out how to get MotionEye to work with two NIC's that is the issue.  It is a little different to VPN access.   It is not involving port forwarding.  It would involve MotionEye being able to handle two network interfaces directly.
All cameras have static IP's so no need for DHCP on the Camera LAN

CRImier, 2020-06-04:
MotionEye doesn't need to know anything about interfaces, it only knows about IP addresses. When ME connects to an IP address, Linux will check which interfaces this IP is accessible through, then route ME's packets accordingly. What's the output of your `ip route` command Don't forget to censor any public IPs (if you have one) or unrelated interfaces when posting it.

austwhite, 2020-06-04:
I have to run off to other work for now.  I'll get back to you on that and see what I can do.  I might be overthinking this again by sounds of it.

CRImier, 2020-06-04:
I.e. here's my output:

```
default via 192.168.1.254 dev eno1 proto dhcp metric 100 
10.8.0.0/24 dev tun0 proto kernel scope link src 10.8.0.19 
169.254.0.0/16 dev eno1 scope link metric 1000 
192.168.1.0/24 dev eno1 proto kernel scope link src 192.168.1.105 metric 100
```
This means that:
1. Packets for 192.168.1.* will go through eno1 (line 4)
2. Packets for 169.254.*.* will go through eno1 too (line 3)
3. Packets for 10.8.0.* will go through tun0 (line 2)
4. any other packets (default) will go through eno1 (line 1)

CRImier, 2020-06-04:
Here's how the output with one additional NIC would look when configured correctly, I imagine:

```
default via 192.168.1.254 dev eno1 proto dhcp metric 100 
10.8.0.0/24 dev tun0 proto kernel scope link src 10.8.0.19 
169.254.0.0/16 dev eno1 scope link metric 1000 
192.168.1.0/24 dev eno1 proto kernel scope link src 192.168.1.105 metric 100
192.168.2.0/24 dev eno2 proto kernel scope link src 192.168.2.1 metric 100
```

joshhodgson, 2020-06-04:
As long as the two networks don't have overlapping IP address ranges, there is **no** config change needed

closed 2021-03-05 by starbasessd