motioneye-project/motioneye #1790: unable to open video device and  connection to camera lost

bedeschenko, 2020-06-09:
hello all i have motioeye running on my rapsberry pi with raspbian buster. i have 3 webcams connected to it. 1 of the webcams work perfectly, while the other 2 are not
the first one with a problem shows  connection to camera lost
while the second one says unable to connect to video device. pls i will be happy if any can help.

TwinBAdams, 2020-06-11:
I've been having the same issue. I thought it was my internet connection and basically gave up on it. Then at the last minute I decided to reduced the frame rates to 15 and the streaming quality to 60, that helped. Running a Pi4 with 2 USB webcams and one OctoPi cam linked in.

bedeschenko, 2020-06-11:
Thank you for your response 
I am sorry i am not by my setup now to try it but i will definitely do and get back to you

bedeschenko, 2020-06-16:
hey Adams thank you for the response. 
but it did not work for me. after doing that i get a green screen no image.

TwinBAdams, 2020-06-16:
Did you do it for each of the cameras individually? Mine have been running since I adjusted those values. Sorry if not the case for you.

closed 2021-03-05 by starbasessd