motioneye-project/motioneye #1795: How to access Motioneye remotely

lodgemoorlad, 2020-06-12:
Hi All
Can anybody tell me if Motioneye can be accessed remotely? 

Seems like most commercial security camera's can,  I have a commercial camera and I can do it with that and it would be amazing if I could do it with this. I guess its not the same format, but for now just asking as it would have amazing potential if so. Thanks.

zagrim, 2020-06-13:
It depends a bit on what you actually mean by remote access. If what you mean is to access the MotionEye GUI (with camera streams and settings etc) outside of your local network, the answer is sure, but you need to set things up yourself (configuring up firewalls, proxies, VPN tunnels, whatever is needed to fulfil your requirements. There have been several issues discussing that scenario already, but the specifics really depend on your requirements.

What MotionEye doesn't provide is similar "magical" remote access via a mobile app that many consumer grade security cameras offer by sending the camera feed to a cloud where you don't actually know who has access to the feed, or the security level of the service.

lodgemoorlad, 2020-06-13:
Hi Elsa

Thanks for getting back to me.

I was hoping it was going to be easy, but expecting something like your
answer.

Ah well, I still love it as it is, it's in a nesting box and I just wanted
to stream it to family and friends who might be interested in what's going
on in there.

But, thanks for taking time out to get back to me, it's much appreciated as
it's saved me a lot of time and pain..

Thanks again

Alex

zagrim, 2020-06-14:
At least these issues relate to what you're asking for, maybe they have something that might help you: #1567, #1365, #669. There's also an article on [how to configure nginx for a proxy](https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx) in the Wiki section which is one option you might use along the way to make MotionEye (or just the stream URLs) reachable from the Internet side.

lodgemoorlad, 2020-06-14:
Hey
Thanks again , seriously kind of you.
I'm out at the moment, but will certainly take a look at the info you have
sent as soon as I get back home
Can't wait.
Thanks
Alex

MYeager1967, 2020-06-16:
I'm using a nginx reverse proxy and it's wonderful. I set the app up on the wife's (and my) phone and all you have to do is open the app.

lodgemoorlad, 2020-06-16:
Hi , Myeager

To be honest, I've never heard of this, do you have any idea where I could
find out more, thanks
A

MYeager1967, 2020-06-16:
It's in the wiki - https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx 
I'm running a Synology NAS on my network and nginx already runs on that, so it was super easy for me. I was in the process of setting it up when I realized I already had it though and it was going pretty good.

lodgemoorlad, 2020-06-16:
Hey
Big thanks for this, can't believe I've never come across this before
Will check it out as soon as I get home, thanks appreciated.👍
A

closed 2021-03-05 by starbasessd

ricso, 2024-02-10:
> It's in the wiki - https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx I'm running a Synology NAS on my network and nginx already runs on that, so it was super easy for me. I was in the process of setting it up when I realized I already had it though and it was going pretty good. Passes the wife acceptance test with flying colors and I enjoy having easy and secure access to it as well.

@MYeager1967  Hey Michael, I hope all is well with you.  I know it's been quite a while since you posted this comment but would you happen to have (or could cook-up) the steps you used to setup MotionEye access via NGINX on your Synology NAS?  I found a video on how to setup NGINX on my Synology NAS but it seems pretty complicated (for me) and I couldn't figure out how to adapt the instruction on the motioneye wiki to it.  Thanks in advance!