motioneye-project/motioneye #1799: Sequential filenames

unicornrobot, 2020-06-14:
Is it possible to create sequential filenames in the still image settings?
I've tried :
folder name/filename_%03d 
Which i'd hoped would create a 3 digit incremental filename, but it doesn't.
 I need the files to be in a single folder for importing into a video editing software for timelapse. 
Sorry if this is documented/asked before, I couldn't find the info . Many thanks.

sethtipton, 2020-08-29:
I'd also like to know how to do this.

sethtipton, 2020-08-29:
Here's a list of available variables, although it doesn't include a sequence number
https://motion-project.github.io/motion_config.html#OptTopic_Image_Processing

If you're on a mac, I found this helpful
https://tidbits.com/2018/06/28/macos-hidden-treasures-batch-rename-items-in-the-finder/

lucianaohira, 2020-12-01:
Hey, hi! 
In the field "Image File Name", we just wrote %v for saving the still images at a number sequence like 01.jpg, 02.jpg and so on.

starbasessd, 2020-12-01:
You can point at the tool tip for file name (blue question mark) to get available filename variables.

lucianaohira, 2020-12-01:
> question

OMG!!!! I was clicking at the blue question mark instead of just pointing... how newbie... Thanks @starbasessd !!!

unicornrobot, 2021-03-04:
> Hey, hi!
> In the field "Image File Name", we just wrote %v for saving the still images at a number sequence like 01.jpg, 02.jpg and so on.

Thanks. But this overwrites the image, not creates a new one.

starbasessd, 2021-03-04:
@unicornrobot if you use something like
%Y-%m-%d-%H-%M-%S-%v
or
%Y-%m-%d-%H-%M-%S-%q
you will get individual filenames with the unique endings.

unicornrobot, 2021-03-04:
thanks again @starbasessd but this approach does still not create sequential images with unique endings. they always end in "01.jpg" using %v and always "00.jpg" using %q
What am I missing here?!

starbasessd, 2021-03-04:
Are you using motion triggered or intervals? If intervals, it doesn't keep
track of the %v or %q numbers.
I don't know if this might help:
https://motion-project.github.io/motion_config.html#conversion_specifiers
The one you could test would be:
%{dbeventid}

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

unicornrobot, 2021-03-04:
I'm using intervals. I understand now that intervals does not keep track of events(%v) or frames(%q)
I do find it odd that there isnt an easy way to capture a simple 001..002..003..etc sequence though.
I've looked at %{dbeventid} but it doesnt seem to be able to generate sequential numbers.
Thanks for your help.

starbasessd, 2021-03-04:
You could probably do something like this (but I'm not a programer)
variable: image_number
check for new file
rename file to image_number
increment image_number
jump back to check for new file
I could probably do something like that in python or bash, and run it as a
crontab job (depending on the interval)
Might be something to try if you want to learn python or bash. I do a few
of my own python scripts and bash scripts...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JamBalaya56562, 2026-11-08:
Closing with an answer: Motion's file name specifiers have no counter for interval snapshots, as found above, as `%v` and `%q` count events and frames. For a timelapse, motionEye can make the movie itself: in the picture browser, the timelapse button of a folder joins its pictures in time order, so no sequential names are needed.

closed 2026-11-08 by JamBalaya56562