motioneye-project/motioneye #1801: How to fix this fake detection ?

Saku241, 2020-06-16:
Hello 
Sometimes I got my camera which detect fake mouvement and I’d like to fix it, I try multiple things but still the same

https://drive.google.com/drive/folders/1-8Ya_qDtP_HwlrB7xYXfKQbMtui0GveP

You will find one video with its debug files, I could add more if needed

Thanks

Dvorf, 2020-06-18:
this is not motioneye issue, all motion detection question you should address to motion project

zagrim, 2020-06-20:
@Saku241 There seems to be quite a lot of noise in the video. It's quite visible from the original video, with plenty of pixel color changes in the tile wall. You could try increasing "Noise Level" under "Motion Detection" until you no longer get "movement" in areas with actually nothing happening. Alternatively you could "Auto Noise Detection" in the same place.

zagrim, 2020-06-20:
After getting the noise propely dealt with, you'll most likely need to re-adjust the motion detection threshold to get real movenent detected in the way you wish.

Saku241, 2020-06-20:
Alright thanks @zagrim !

closed 2021-03-05 by starbasessd