motioneye-project/motioneye #1802: Failed to upload Media file

jschoo, 2020-06-16:
I set up an upload service to upload media files. But it failed.
Log attached.

   ERROR: failed to upload file "/share/motioneye/Camera3/GT4Floors-2020-06-16/18-03-44.jpg" with service sftp: (48, '')
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
    service.upload_file(target_dir, filename, camera_name)
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
    self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 989, in upload_data
    conn = self._get_conn(filename)
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1033, in _get_conn
    self._conn.setopt(self._conn.SSH_AUTH_TYPES, auth_types[auth_type])
error: (48, '')

How can I solve it?

closed 2021-03-05 by starbasessd

oregling, 2023-03-06:
Hi,

i am facing the exact same issue here with motionEye 0.18.0 on an home assistant OS 2023.3.1.
How did you solve this issue?

MichaIng, 2023-04-11:
Please try the `dev` branch: https://github.com/motioneye-project/motioneye/tree/dev#installation

The HA addon AFAIK uses still uses the Python 2 branch (should be motionEye 0.42.1, actually). There were quite some bugs (due to external service changes) solved in dev, e.g. when uploading to Dropbox: #2396

leochras, 2023-04-11:
As I am running the motioneye addon on my HA instance, I am unable to test it. But lmk when there's an update..

MichaIng, 2023-04-16:
As far as I know, the web API has not changed, i.e. it should be backwards compatible. But there have been a lot of changes which I did not follow, and even quite some between the 0.42.1 release and the latest Python 2 branch commit, so in the end it needs to be tested.

Okay, but then I understand the procedure being motionEye release => 3. => 2. => 1. (while some parallel testing/development with unreleased states can accelerate things, of course).

dermotduffy, 2023-04-16:
> Okay, but then I understand the procedure being motionEye release => 3. => 2. => 1. (while some parallel testing/development with unreleased states can accelerate things, of course).

Exactly. Is there a docker build by any chance available somewhere and perhaps I could start experimenting with 3 at least?

MichaIng, 2023-04-16:
Docker images are here: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye