motioneye-project/motioneye #1807: Can't disable motion detection in docker

think-free, 2020-06-19:
motioneye_1  |    ERROR: failed to disable motion detection for camera with id 1: Failed to connect to 127.0.0.1 port 7999: Connection refused
motioneye_1  |    ERROR: failed to disable motion detection for camera with id 4: Failed to connect to 127.0.0.1 port 7999: Connection refused
motioneye_1  |    ERROR: failed to disable motion detection for camera with id 1: Failed to connect to 127.0.0.1 port 7999: Connection refused
motioneye_1  |    ERROR: failed to disable motion detection for camera with id 4: Failed to connect to 127.0.0.1 port 7999: Connection refused
motioneye_1  |    ERROR: failed to disable motion detection for camera with id 1: Failed to connect to 127.0.0.1 port 7999: Connection refused
motioneye_1  |    ERROR: failed to disable motion detection for camera with id 4: Failed to connect to 127.0.0.1 port 7999: Connection refused

smhgit, 2020-10-26:
@think-free did you ever get an answer? did you find a solution?

think-free, 2020-10-28:
> @think-free did you ever get an answer? did you find a solution?

Disabling camera completely ... 
I think that if you enter the container you can make a (web) request from inside but I didn't investigate too much

closed 2021-03-05 by starbasessd

starbasessd, 2021-03-05:
@think-free Last activity was Oct 28 2020, no further responses. Just trying to clean up issues older than 60 days (Pre 2020-01-01 last activity)
If you still need help, you should be able to re-open.