motioneye-project/motioneye #1808: Ftp Upload / Weathercam / Weatherunderground / Help

mutsched, 2020-06-19:
Hello to all maybe anyone can help me to setup my ftp upload to weatherunderground…
I have build an weather cam and i want to upload pictures fromm my RPI to my Weatherunderground account. I installed Noobs with Raspian, then i installed motioneye and write and start scrit so that it boots up automaticly. I also updatet all and write wpa.supplicant.conf. When i write ip in my windows browser i see image and can adjust the software. As i write above i want tu upload images. So i make an webcam device on WU and get the FTP login Data. I write all in motioneye FTP settings and the „test service“ also shows „Accessing the upload service succeded“. I also loked in the storage and the picture i made is safed as image.jpeg like wu needs… Maybe anyone can help to get this work. Thanks Daniel

[image]

zagrim, 2020-06-20:
I'm sorry, but what is the actual issue you are having? I'm guessing the images aren't somehow going where they should be, but I don't see any description of a failure here :)
Anyway, MotionEye saves the images in a subfolder/subdirectory (you have "Include Subfolders" enabled) similar to how it saves the locally. Maybe the directory structure and/or file naming isn't what WU expects? WU seems to require logging in in order to see the uploading instructions, something I don't wish to do myself, so this I'm just guessing here.

mutsched, 2020-06-20:
Hello and thanks for answer and your time.
When i write my account login (Device ID/Password) to motioneye and i test service i get answer "Accessing the upload service succeded"
When i change password or when i change my Device ID i get error message so my login is correct. 
Also when i change port number i get also error so connection seems to be ok.
The pictures are safed on /var/lib/motioneye on my RPI.
I changed also filename from Date(year/month/day/min/sek) to the name image. So the safed picture is named image.jpg and it will be override a every new image i take picture with cam.
But when i test service (button in motioneye) there will be no picture upload....
Also my Webcam device on WU shows offline.. In never go online at my tests.
Does anyone now if i make a special code that every minute motioney uploads the image...
Or what i have to do do get my pictures to Weatherunderground ??
I searched the internet but i found no resolution... I also tryed to build an script with that i can upload directly the image.jpg from my storage folder to WU but also does not work for me... maybe becouse of my bad coding skills.... If i dont get it to work mybe somebody knows a other service/app/os what works at picture upload to wu... Thanks

mutsched, 2020-06-20:
Mybe i dont have permissions to upload file from that /var/lib/motioneye folder...
I tryed on Raspbian Desktop to copy my /var/lib/motioneye/image.jpg but dont work because no permission... How can i control this ?

mutsched, 2020-06-20:
After long time of try i get this message in log file:
[image]
So my upload service per FTP to weatherunderground should work...
Maybe problem on WU ?!

zagrim, 2020-06-21:
I don't think the "Test Service" button uploads any image, it just checks that authentication works and that it can create a new directory and/or some generic file on the service, and then it deletes those. So basically the upload should work as your log also seems to show.

I have no idea about how WU works, but I guess they have some sort of support forum where you could as about that. 
Note that you could also use a FTP client to simulate the image upload (connect to WU using those credentials and upload an image there) to see if that makes any difference. That way you could also easily test different file name schemes and/or directory structures if that turned out to be relevant.

gnattycole, 2021-02-01:
Hi, Did you ever get this fixed? I'm looking at doing the same if possible.

JamBalaya56562, 2026-11-08:
Closing, as the upload itself worked, as your log showed, so what remains is the format Weather Underground expects. motionEye uploads each file under its own name, and with "Include Subfolders" also in the same subfolders as locally. A service that wants one fixed file name needs a script instead, run with "Run a Command" after each picture. Test Service only checks that it can log in and write a test file.

closed 2026-11-08 by JamBalaya56562