motioneye-project/motioneye #1809: ARM64 Docker hub builds

vdo, 2020-06-20:
Right now RPi 4 is able to run 64 bit OS, like ubuntu, and official images will be added too soon.

I think it should be easy to add the arm64/aarch64 builds in docker hub, just like armhf

minnixtx, 2020-08-30:
I was about to write a feature request for this as well. Currently running Ubuntu server 64 bit on RPi 4. With the RPi 8GB version out now it will be useful for processes to be able to address more RAM than the current 2GB on 32 bit OS.

kleini, 2020-08-30:
It is very hard to add a ARM64 image. First, these images are created using Travis CI, which only provides x86 hardware. Running other architectures is always then a problem. I could implement this using a container image for multiarch support. Maybe it supports ARM64, too, but this needs to be checked. Furthermore it requires an according base image with ARM64 architecture.

If you are so keen, to have this supported, feel free to create a pull request bringing the necessary changes into the project.

vdo, 2020-08-30:
It's not that hard using HW emulation and the buildx feature of docker. 
See:
https://www.docker.com/blog/multi-arch-build-and-images-the-simple-way/

kleini, 2020-08-30:
@vdo Then go ahead and provide the pull request.

closed 2021-03-05 by starbasessd