motioneye-project/motioneye #1814: Camera drops out and stops working constantly

Planeguy102, 2020-06-26:
I'm having a very annoying issue with one of my cameras. When I create a new install of Motioneye OS (using a Raspi 4, plugged into the official power supply, running 20190911), all seems fine. But as I start modifying settings, one section after another, at some point the camera will stop working, forcing me to reboot the system even if I changed a very minor setting (Most recent example is when I changed the size of the text.) After I had everything configured, though, with no dropouts, the camera would just turn off after about 20-30 minutes, again forcing a reboot. To top it off, it always sets the time 5 hours ahead of what it's supposed to, no matter if I use HTTP or NTP or different servers for both. I'm honestly at a loss for what could be causing it.

merrycachemiss, 2020-06-27:
I've noticed the same thing happen (the permanent-until-reboot dropout part) after a bunch of config changes in a row, and I'm not running Motioneye OS, just the regular install on Raspbian, latest release version as of today's date. Using picam v2 on a pi4 4GB, official power supply like you.

I might not be remembering this correctly, but I think the streaming URL still showed the live feed during the dropout -- it was just the config/dashboard page that didn't show video. What about you?

Since I just got this, I'm still doing the setup, and I'm hoping it doesn't dropout when things are ready for actual use -- does this happen for you when you don't do any configuration stuff after a fresh boot? I'm not at a place yet where I can stop config changes and just have it running, as I want to get things in a more secure state.

Planeguy102, 2020-06-27:
From what I can gather, it's because of my Samba fileshare server. As soon as I switch the file storage back over to internal, the camera comes right back. This happened to me in both 20190911 and 202006.

closed 2021-03-05 by starbasessd