motioneye-project/motioneye #1819: MotionEye runs in docker as user root.

marcopaganini, 2020-06-30:
Motioneye runs as root in docker. This is a bad practice as it may lead to privilege escalations in case of security issues. Is there any specific issue why we don't run docker as a regular user? Please indicate if this is a desirable development and I'll start working on a PR to fix it.

kleini, 2020-06-30:
Running services in a container as root is a bad practice? Where is that written down? The container is exactly meant for running services as root without having the problem of any privilege escalations as the service is locked down into a container. If Docker has issues, than it is a Docker issue and not a Linux container issue. Please distinguish that as Docker is not **the** only solution any more for running container images.

zagrim, 2020-07-01:
Root in docker container is way different than root in a VM, i.e. Docker is bad at separating things running inside the container from this running on the host. https://security.stackexchange.com/a/176350/67048 and https://neoteric.eu/blog/docker-containers-with-root-privileges/ mention some downsides of running things as root in containers. Not all might apply in typical ME setup but nevertheless there is a point to be considered.

kleini, 2020-07-01:
All your links talk about malicious implementations of container runtimes and wrong usage of containers. There is exactly no risk to run processes as root in containers except container runtime is buggy or user does a mistake. And for the Linux kernel there is no possibility for the root process in the container to excape as it is a fake root. So, there is no need to run processes in containers not as root.

closed 2020-07-01 by kleini

marcopaganini, 2020-07-01:
Hmmm, I partially agree. It is very possible to escape a chroot (fake root)
*if* you're root inside the jail. In docker/modern kernels, I believe this
problem has been eliminated by namespaces (someone, please correct me if
I'm wrong) and we'd be protected (even running as root), unless a new
exploit is discovered. In an ideal world, running as root should not be a
problem, but when it comes to security I'm a "belt & suspenders" type of
person. In general, having yet another service running as root (when it
could be easily avoided) may not be the best decision.

Out of curiosity: It shouldn't be so hard to change the Dockerfile so we
can easily build an image that runs motion as root or any other
unprivileged user. Would you be against this approach? Ideally, by default
it would run as root, but building the container with a special arg would
cause it to run as any other user.

Thoughts?

kleini, 2020-07-01:
You're free to change the UID in a PR. Go ahead, everything, that is contributed to this project does not need to be solved by the owner, right?

marcopaganini, 2020-07-01:
Oh, no problem at all. I'm actually offering help. Just trying to see if
the owner approves of the general direction of the PR before I take some
time to write it :)

PR #1825 (2020-07-02): Add support to run motionEye as a non-root user.
- Added RUN_UID and RUN_GID args to Dockerfile. This allows the creation
  of an image that runs as a non-privileged user. If no args are
  specified at the container creation time, UID and GID will default
  to root/root, as before.
- Reordered commands in the Dockerfile a bit.
- Removed /var/run/motion as a PID, since we may want to run
  another instance of motion on a docker and this could prevent it.
- Added docker specific documentation.

Fixes: #1819