motioneye-project/motioneye #1826: Web Interface has no options

japananman, 2020-07-02:
Setting up MotionEye on top of Raspian Buster Full. 
I am able to access the web interface, but have no camera and nothing happens if I click on either the User or Menu buttons. 
I have very little experience in Pi, this is my first venture down the rabbit hole, so any assistance would be appreciated. 

[image]

zagrim, 2020-07-02:
Please check for any errors on the RPi, the explanation is likely there (`journalctl -u motioneye` will probably work).

japananman, 2020-07-02:
-- Logs begin at Thu 2020-07-02 04:17:01 BST, end at Thu 2020-07-02 14:59:39 BST. --
Jul 02 14:54:46 raspberrypi systemd[1]: Started motionEye Server.
Jul 02 14:54:48 raspberrypi meyectl[328]:     INFO: hello! this is motionEye server 0.42.1
Jul 02 14:54:50 raspberrypi meyectl[328]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Jul 02 14:54:50 raspberrypi meyectl[328]:     INFO: cleanup started
Jul 02 14:54:50 raspberrypi meyectl[328]:     INFO: wsswitch started
Jul 02 14:54:50 raspberrypi meyectl[328]:     INFO: tasks started
Jul 02 14:54:50 raspberrypi meyectl[328]:     INFO: mjpg client garbage collector started
Jul 02 14:54:50 raspberrypi meyectl[328]:     INFO: server started
Jul 02 14:54:50 raspberrypi meyectl[328]:     INFO: mjpg client garbage collector started
Jul 02 14:54:50 raspberrypi meyectl[328]:     INFO: server started

japananman, 2020-07-03:
> Please check for any errors on the RPi, the explanation is likely there (`journalctl -u motioneye` will probably work).

Jul 02 15:24:52 raspberrypi meyectl[1815]: INFO: hello! this is motionEye server 0.42.1
Jul 02 15:24:53 raspberrypi meyectl[1815]: INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Jul 02 15:24:53 raspberrypi meyectl[1815]: INFO: cleanup started
Jul 02 15:24:53 raspberrypi meyectl[1815]: INFO: wsswitch started
Jul 02 15:24:53 raspberrypi meyectl[1815]: INFO: tasks started
Jul 02 15:24:53 raspberrypi meyectl[1815]: INFO: mjpg client garbage collector started
Jul 02 15:24:53 raspberrypi meyectl[1815]: INFO: server started
lines 12-38/38 (END)

Motion will capture a video, but I can't add a camera to motioneye. I have initialized it in raspi-config, but no user login or camera is available in the web interface, still.

zagrim, 2020-07-03:
Did you log in as "user" or "admin"? Admin is required for adding cameras, at least. Although I didn't myself succeed to log in as "user" in a setup with no custom configuration done, ended up back at the login dialog instead.

Please also check the browser console for errors (available via developer tools or whatever it is called in your browser of choice).

> I can't add a camera to motioneye. I have initialized it in raspi-config, but no user login or camera is available in the web interface, still.

You have initialized Motioneye in raspi-config, or did I read that wrong? Anyway, I'm running ME on Raspbian Stretch and don't see any connection with ME and things available in raspi-config.

japananman, 2020-07-03:
@zagrim Thank you for helping me with this, I'm at a loss.

> Did you log in as "user" or "admin"? Admin is required for adding cameras, at least. Although I didn't myself succeed to log in as "user" in a setup with no custom configuration done, ended up back at the login dialog instead.
There is no login prompt when I get to the web interface.

> Please also check the browser console for errors (available via developer tools or whatever it is called in your browser of choice).
I'm getting this in the browser console (I don't understand this AT ALL)

Uncaught SyntaxError: Unexpected end of input

jquery.timepicker.min.js?v=0.42.1:1 Uncaught ReferenceError: jQuery is not defined
    at jquery.timepicker.min.js?v=0.42.1:1
    at jquery.timepicker.min.js?v=0.42.1:1
jquery.mousewheel.js?v=0.42.1:22 Uncaught ReferenceError: jQuery is not defined
    at jquery.mousewheel.js?v=0.42.1:22
    at jquery.mousewheel.js?v=0.42.1:24
main.js?v=0.42.1:5182 Uncaught ReferenceError: $ is not defined
    at main.js?v=0.42.1:5182
DevTools failed to load SourceMap: Could not load content for chrome-extension://gighmmpiobklfepjocnamgkkbiglidom/include.preload.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME 

> 
> You have initialized Motioneye in raspi-config, or did I read that wrong? Anyway, I'm running ME on Raspbian Stretch and don't see any connection with ME and things available in raspi-config.

I initialized the camera, under device peripherals.

zagrim, 2020-07-03:
The errors in the browser console hint that for some odd reason the jQuery library (which is bundled in to the MotionEye GUI) is not being found. Since it is present locally and not loaded over the network I find it hard to figure out any plausible reason for it not being found.
Do you by chance have some browser extension in Chrome that might block some Javascript files from being loaded (for example, if it thinks they are threat to security)? Could you please try using another browser if you have such installed, or disabling Chrome extensions (using incognito mode might accomplish that, too)?

japananman, 2020-07-03:
@zagrim I did open it up in Opera and could actually login! But...
Now I am getting no image on the camera. 

my (journalctl -u motioneye leaves me with this: 10k plus lines of this:
> Jul 03 10:33:20 raspberrypi motion[1270]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Jul 03 10:33:20 raspberrypi motion[1270]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Jul 03 10:33:20 raspberrypi motion[1270]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Jul 03 10:33:20 raspberrypi motion[1270]: [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
Jul 03 10:33:20 raspberrypi motion[1270]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Jul 03 10:33:21 raspberrypi motion[1270]: [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
Jul 03 10:33:21 raspberrypi motion[1270]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Jul 03 10:33:21 raspberrypi motion[1270]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Jul 03 10:33:21 raspberrypi motion[1270]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Jul 03 10:33:30 raspberrypi motion[1270]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Jul 03 10:33:30 raspberrypi motion[1270]: [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
Jul 03 10:33:30 raspberrypi motion[1270]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Jul 03 10:33:30 raspberrypi motion[1270]: [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
Jul 03 10:33:30 raspberrypi motion[1270]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open

zagrim, 2020-07-03:
Great if it works with Opera (doesn't quite explain the original issue but maybe you'll figure it out eventually) :)

So the camera you have is a Pi Camera? Have you checked everything mentioned in https://github.com/ccrisan/motioneye/wiki/Troubleshooting#raspberrypi-camera-module-not-detected ? I've not ever configured a Pi Cam and can't give any specific advice, but you might find something searching earlier issues here since that general topic rises up frequently.

zagrim, 2020-07-04:
#1645 seems to have had the same errors, but doesn't detail the actual fix (other than that checking again the camera configuration and/or re-adding the camera fixed the issue).

closed 2021-03-05 by starbasessd