motioneye-project/motioneye #1829: MotioneyeOS cannot play its own files

gzcwnk, 2020-07-04:
"the video cannot be played as it is corrupt"  I had this working on some cameras (but not others) and now none work. I can download and play the mp4 in vlc OK.   This is a docker image,

root@debian-docker-001:~# docker ps
CONTAINER ID        IMAGE                            COMMAND                  CREATED             STATUS              PORTS                                                                                                                         NAMES
c6055b49c9ab        ccrisan/motioneye:master-amd64   "/bin/sh -c 'test -e…"   4 weeks ago         Up 7 days           0.0.0.0:8765->87                          65/tcp, 0.0.0.0:18081->8081/tcp, 0.0.0.0:18082->8082/tcp, 0.0.0.0:18083->8083/tcp   motioneye
root@debian-docker-001:~#

I have been altering some of the quality settings so I suspect this might be the issue?   Basically I moved the quality setting in movies from 75% to 100%.

I went back to some original files and they play fine, so am a bit stumped.

This is my first ever container so I'm a bit clueless on them.

=======
root@debian-docker-001:/var/lib/motioneye# docker info
Containers: 1
 Running: 1
 Paused: 0
 Stopped: 0
Images: 2
Server Version: 18.09.1
Storage Driver: overlay2
 Backing Filesystem: extfs
 Supports d_type: true
 Native Overlay Diff: true
Logging Driver: json-file
Cgroup Driver: cgroupfs
Plugins:
 Volume: local
 Network: bridge host macvlan null overlay
 Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
Swarm: inactive
Runtimes: runc
Default Runtime: runc
Init Binary: docker-init
containerd version: 9754871865f7fe2f4e74d43e2fc7ccd237edcbce
runc version: 1.0.0~rc6+dfsg1-3
init version: v0.18.0 (expected: fec3683b971d9c3ef73f284f176672c44b448662)
Security Options:
 apparmor
 seccomp
  Profile: default
Kernel Version: 4.19.0-8-amd64
Operating System: Debian GNU/Linux 10 (buster)
OSType: linux
Architecture: x86_64
CPUs: 2
Total Memory: 3.854GiB
Name: debian-docker-001
ID: CDOG:WSJW:2HFO:QT3P:EUAU:GB2Q:YJWN:MB66:34VS:MVSM:YUCY:MRLT
Docker Root Dir: /var/lib/docker
Debug Mode (client): false
Debug Mode (server): false
Registry: https://index.docker.io/v1/
Labels:
Experimental: false
Insecure Registries:
 127.0.0.0/8
Live Restore Enabled: false

WARNING: No swap limit support
root@debian-docker-001:/var/lib/motioneye#
=======

=======
root@debian-docker-001:/var/lib/motioneye# docker image inspect ccrisan/motioneye:master-amd64
[
    {
        "Id": "sha256:692eccfe45a604d8e5331274c2388a2ecf65bb3f2f033232ed7e38f6f43cd1c3",
        "RepoTags": [
            "ccrisan/motioneye:master-amd64"
        ],
        "RepoDigests": [
            "ccrisan/motioneye@sha256:1704cddc44e0bc7e1c4aff9f1f6546127584e8351c04a0faa1d46c7fb791180b"
        ],
        "Parent": "",
        "Comment": "",
        "Created": "2019-12-15T12:00:16.684380673Z",
        "Container": "09da85ebd01081f7ea6eb685819722d34ea91b9ef7d6e7b6c91640c5bac22166",
        "ContainerConfig": {
            "Hostname": "09da85ebd010",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "ExposedPorts": {
                "8765/tcp": {}
            },
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
            ],
            "Cmd": [
                "/bin/sh",
                "-c",
                "#(nop) ",
                "EXPOSE 8765/tcp"
            ],
            "ArgsEscaped": true,
            "Image": "sha256:472e8eda4d05c12813d9eefd16b3dbcbcfa7394b2339bf932d97f09ce4be4109",
            "Volumes": {
                "/etc/motioneye": {},
                "/var/lib/motioneye": {},
                "/var/run/motion": {}
            },
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": [],
            "Labels": {
                "maintainer": "Marcus Klein <himself@kleini.org>",
                "org.label-schema.build-date": "2019-12-15T11:59:13Z",
                "org.label-schema.docker.dockerfile": "extra/Dockerfile",
                "org.label-schema.license": "GPLv3",
                "org.label-schema.name": "motioneye",
                "org.label-schema.url": "https://github.com/ccrisan/motioneye/wiki",
                "org.label-schema.vcs-ref": "9f667cd487b141506635322e84a09355a0295857",
                "org.label-schema.vcs-type": "Git",
                "org.label-schema.vcs-url": "https://github.com/ccrisan/motioneye.git"
            }
        },
        "DockerVersion": "17.09.0-ce",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "ExposedPorts": {
                "8765/tcp": {}
            },
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
            ],
            "Cmd": [
                "/bin/sh",
                "-c",
                "test -e /etc/motioneye/motioneye.conf ||     cp /usr/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf ;     /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf"
            ],
            "ArgsEscaped": true,
            "Image": "sha256:472e8eda4d05c12813d9eefd16b3dbcbcfa7394b2339bf932d97f09ce4be4109",
            "Volumes": {
                "/etc/motioneye": {},
                "/var/lib/motioneye": {},
                "/var/run/motion": {}
            },
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": [],
            "Labels": {
                "maintainer": "Marcus Klein <himself@kleini.org>",
                "org.label-schema.build-date": "2019-12-15T11:59:13Z",
                "org.label-schema.docker.dockerfile": "extra/Dockerfile",
                "org.label-schema.license": "GPLv3",
                "org.label-schema.name": "motioneye",
                "org.label-schema.url": "https://github.com/ccrisan/motioneye/wiki",
                "org.label-schema.vcs-ref": "9f667cd487b141506635322e84a09355a0295857",
                "org.label-schema.vcs-type": "Git",
                "org.label-schema.vcs-url": "https://github.com/ccrisan/motioneye.git"
            }
        },
        "Architecture": "amd64",
        "Os": "linux",
        "Size": 410501519,
        "VirtualSize": 410501519,
        "GraphDriver": {
            "Data": {
                "LowerDir": "/var/lib/docker/overlay2/622b0a75b885d44058fa20cf78c3c8df3a7ce0054fd23eb8bfee375ad8cf6a0c/diff:/var/lib/docker/overlay2/64664c9ef35f7e6eab9f9af82fcf1b4a876b86dca8ff0560ac7e1e621f7173a2/diff:/var/lib/docker/overlay2/ed9d79b689408425920b467cfc99c601d780a7aca987077631f89e41e050483e/diff:/var/lib/docker/overlay2/e5bbb0b6a7cac5d1cc1588c99c4fc3568df788b6ebf5b9efdf0af69b160174ab/diff:/var/lib/docker/overlay2/423d2ebb612b67f3cb71dec3fd8704345166d0568c554b81a552b4254a124ac6/diff:/var/lib/docker/overlay2/2568040f17b68c4a1683eebcea28c49871e0f9db03f010544a5073536b836153/diff",
                "MergedDir": "/var/lib/docker/overlay2/c7ffb526adc7a572c0662e21138a04c51a4cad676681307cfe9d47324c6322cf/merged",
                "UpperDir": "/var/lib/docker/overlay2/c7ffb526adc7a572c0662e21138a04c51a4cad676681307cfe9d47324c6322cf/diff",
                "WorkDir": "/var/lib/docker/overlay2/c7ffb526adc7a572c0662e21138a04c51a4cad676681307cfe9d47324c6322cf/work"
            },
            "Name": "overlay2"
        },
        "RootFS": {
            "Type": "layers",
            "Layers": [
                "sha256:d33f208862dd35ea0d843ea338eedf2b5504220e0bc9aa2275a86f97241b2d9b",
                "sha256:7027b2a028fc02466d63aefcca83a1f7e457dcd7c8be434c1d9933c17633e3bc",
                "sha256:2181ca10496e8fcdb10a2ff5fe0af0e6f4616beecce178f6fd6b499a14083e1a",
                "sha256:e338c474bd3b9f2dd78c2970dfe0a3b5f7df11886fccb02579911bfa5c0e8823",
                "sha256:5916eb004b2a33e38f49776c35bb6ac41100f7e156288c9770a6083aef836f81",
                "sha256:73e2f7e8499c65968e311ef584abea865c151d888808326c99f49220bd0ec883",
                "sha256:315b9edc69693b7d7ac320301a9120407eb79d6be44c391a0da92b1c4a1dea86"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
root@debian-docker-001:/var/lib/motioneye#
==========

gzcwnk, 2020-07-04:
I also dont know how github page syntax works, LOL. oops

gzcwnk, 2020-07-04:
Ok I can fix / reproduce this by simply changing the quality from 100% back to 75% I think.  Will do some more testing.

gzcwnk, 2020-07-04:
Ok I can confirm that going from the std 75% to 100% causes some sort of software issue with playback. Going to 95% is OK.

zagrim, 2020-07-05:
You could probably use `ffprobe -show_streams myvideofile.mp4` to inspect the quality related attributes in the produced videos, but I'd guess that with 100% you get something extreme for "profile" like [High 4:4:4 Predictive](http://blog.mediacoderhq.com/h264-profiles-and-levels/) which probably just isn't supported by the codecs included in MotionEyeOS. The way `ffmpeg` decides the encoding parameters based on what MotionEye passes to Motion (which then uses `ffmpeg` to assemble the videos containing motion) is not clear to me, though.

zagrim, 2020-07-05:
This relates to #1003 and #1824, at least.

davebdb, 2020-07-05:
I opened up ticket 1003.

2boot, 2020-07-13:
Sorry to jump in but I had a similar problem, well maybe remotely similar [https://github.com/ccrisan/motioneyeos/issues/2013](url) was to do with "changed video mode from H.264 to H.264/OMX and videos actually play." This is intended to help but maybe it won't, though worth a check.

SargFowler, 2021-02-07:
I've just left the video passed straight thru rather than encoded and I can now see the video in the web browser and via QuickTime.

hasse-basse, 2022-06-15:
Iphone has constraints when it comes to resolution of mp4, max is 1080p. For me it was as simple as setting the Movie Quality to 75% instead of 100%, et voilà, I could play the movies on my iPhone.

When it comes to Movie Format, the only mp4 that can be used is H.264 (.mp4) The other two H.264 options, with more letters in the name, can often not be used, it seem to be an issue about what OS you are using and what Codecs that comes with the OS. I use Raspberian Buster and H.264 (.mp4) is the only option that works for me. This also affects the Time laps movie making.

JamBalaya56562, 2026-11-10:
Closing, as the replies above found it: at Movie Quality 100% the H.264 movies go beyond what browsers and phones play, while VLC still does; 75% or Movie Passthrough, with the H.264 (.mp4) format, plays in the browser again.

closed 2026-11-10 by JamBalaya56562