motioneye-project/motioneye #1830: Motion won't create more than one video

isaacewing, 2020-07-05:
I've spent way to much time trying to figure this out. 

I'm using a raspberry pi 4 with 128gb sd card. it has the latest version of motioneyeos.

as a test, the options for still images is set to  capture every minute and that works like a champ. i know there are a couple posts already with a similar issue like "the images save but the videos don't" and I tried those but it didn't actually do anything.

now, i've toggled every option  under movies (with a preference for hevc) and yet regardless, it will record the exact movie length requested then it will not record anything else unless restarted. I also toggled each option in movies and bound the recording mode with motion detection and it works to some degree but it has super weird lag issues and I actually need it to record non-stop.

the startup is fine and the only issue i had was that weird one when it complains about the module not being 16 so i changed the resolution of the  video from 1920x1080 and updated it to 1920x1072. although switching the resolution to that prevents the program from giving errors, there still isn't any solution that i could find.

one last weird detail, so i have 16 raspberry pi's setup just like i described, they all worked a couple days and i can see from the videos that were recorded, they all worked like a champ ALL THE WAY up until 12:00 the next day. this has happened in the past too. there's something about the new day that crashes the program.

anyway, hopefully there's something i missed. if so and even if not... any suggestions or ideas. the other day, i had it working so i know it works and yet i can't figure out what happened or why it stopped.

ADYoungberg, 2020-07-08:
I'm having the exact same problem, ended up having to revert to 20180627 to get video recording to work again.

belese, 2020-07-24:
i've got same problem too.

closed 2021-03-05 by starbasessd