motioneye-project/motioneye #1832: Not working RTSP on raspberry pi 4 (Deb 10)

bitleadmin, 2020-07-06:
Hello! I have RP 4 with 4Gb RAM (Deb 10) After install Motion+motioneye 

motionEye Version | 0.42.1
-- | --
Motion Version | 4.3.1
OS Version | Raspbian 10

It can't connect any RTSP cameras - in logs i see

Jul 06 15:36:46 raspberrypi motion[558]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Jul 06 15:36:46 raspberrypi motion[558]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config
Jul 06 15:36:50 raspberrypi motion[558]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Jul 06 15:36:53 raspberrypi motion[558]: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open came
Jul 06 15:36:53 raspberrypi motion[558]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
Jul 06 15:36:56 raspberrypi motion[558]: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open came
Jul 06 15:36:56 raspberrypi motion[558]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
Jul 06 15:36:56 raspberrypi motion[558]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Jul 06 15:36:56 raspberrypi motion[558]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config
Jul 06 15:37:00 raspberrypi motion[558]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera

zagrim, 2020-07-06:
Does the same URL you've entered in MotionEye "Video Device" -> "Camera Device" work in e.g. VLC player?

bitleadmin, 2020-07-06:
> Does the same URL you've entered in MotionEye "Video Device" -> "Camera Device" work in e.g. VLC player?

Yes, working on VLC - stream - rtsp://log:pass@192.168.3.180/videoMain

zagrim, 2020-07-06:
I think in MotionEye you can use that URL as such (leaving the Username and Password fields empty), or using `rtsp://192.168.3.180/videoMain` as the URL and entering username and password in their separate fields when adding the camera. Do either of those work?
Anyway, [doing a search on "RTSP credentials"](https://github.com/ccrisan/motioneye/issues?q=is%3Aissue+rtsp+credentials) here might also give some ideas (might be a specific quirk of the camera model/brand, too).

closed 2021-03-05 by starbasessd