motioneye-project/motioneye #1835: Video is blurry or showing rainbow colors, what could be the issue?

DBuit, 2020-07-07:
Hello,

I got motion eye running with my reolink rlc-410w.
It is detecting motion but i am getting alot of images that are blurry or showing alot of colors like a rainbow:
Example:
[image]

What could be the problem? I have the reolink app running with motion and never seen this kind of images on that...

jasaw, 2020-07-09:
It could be packet drop, or broken camera.
Are you reading the camera stream via UDP or TCP? Try TCP.
Try lowering the resolution and/or framerate too.

antonoal, 2020-07-22:
I've had a similar issue although sporadically. At some point I tracked it down to the following sequence. When I reboot one of my cameras I get that in motioneye.

closed 2020-08-01 by DBuit

1ubuntuuser, 2021-05-11:
I think it's related the reolink "Iframe configureation" issue. Blue iris forums talk of it.