motioneye-project/motioneye #1840: Capture Mode for Still Images and Recording Mode for Movies share the same variable

rhoenen, 2020-07-11:
In my application, I need to record Movies 24/7 and only take Still Images when motion is detected outside the Working Schedule.

Currently, if I change the Capture Mode for Still Images to "Motion Triggered", it also changes the Recording Mode for Movies to "Motion Triggered", and if I change the Recording Mode for Movies back to "Continuous Recording", the Capture Mode for Still Images also changes to "All Frames".

Please separate these two variables so that the software will work for my application.

zagrim, 2020-07-11:
Unfortunately it might not be possible, at least without significant re-design. This is due to MotionEye being dependent on Motion, which is aimed at, well, doing things based on motion detection. Recording mode "Continuous Recording" enabled Motion option `emulate_motion`, which makes it act as if there's motion occurring continuously (`emulate_motion` is originally, I guess, meant as an aid for tuning the configuration, and MotionEye just piggy-backs that functionality for being able to do continuous recording / snapshot capture. So there's a good reason those two options are interconnected. Maybe that fact should somehow be explained in the tooltip text, too.

oofnikj, 2020-11-02:
Yes, I found this confusing too. 
I want to be able to set video to "continuous recording" while capturing stills on an interval for timelapse, but whenever I change still capture mode to interval, video recording jumps back to motion triggered. There doesn't seem to be a way to do this - maybe there is some workaround?

starbasessd, 2020-11-02:
AFAICT, they are mutually exclusive, currently.You can turn a set of still
capture images into a mjpeg video, but I think it has to do with interrupts
and how the video is processed. I don't think it's an option in motion by
itself, either.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-05 by starbasessd