motioneye-project/motioneye #1842: Integration with motion api and status reflection on motioneye

gerasimone, 2020-07-14:
version : latest pip 0.42.1
installed as a package or compiled from sources: yum rpm for motion and pip for motioneye.
standalone or part of third party: standalone
video stream source: net cam rtsp
hardware: x64 
operating system : Centos 7

I don't understand how to use the motion api rest for control cameras and how they reflect on motioneye status.
I have a camera that is turned on only when alarm is activated, so if I leave always enabled in motioneye got a lot of errors in system log, so i try to disable the cam with the motion api rest:
```
curl http://192.168.0.1:7999/3/action/quit
quit in progress ... bye 
Done 
```
and the errors stop flloging in the log, but when the camera restart, i want to restart the motion processing with:
```
curl http://192.168.0.1:7999/3/action/restart
Restart in progress ...
Done 
```
but motioneye never see the changes.
In the same way, if I stop the motion detection with api rest, motioneye never update the input toggle

Could someone explain how to use in the proper way?
thanks

fabiosoft, 2020-07-23:
Same issue, but with motion status START/PAUSE. I think one way is to monitor `/status` endpoint

Ricky0351, 2021-03-10:
hello, I have the same issue (deactivating camera with http request, or deactivating motion detection also with http request, but the status doesn't change in motioneye gui interface). Have you find a solution to solve the problem? thank you

JamBalaya56562, 2026-11-08:
Closing this. Motion's webcontrol acts on Motion only, so motionEye does not see a camera quit or restarted there, nor a pause of motion detection: it reads the detection state from Motion once after Motion starts and then keeps it. To keep the UI in step, make the change through motionEye instead: turn the camera off and on with its switch in the settings, and, from 0.45.0b1, pause and resume detection with the new "Motion detection toggle button" (#3468) or its actions `/action/<camera id>/detectionpause` and `/action/<camera id>/detectionstart`, which update the state motionEye shows.

closed 2026-11-08 by JamBalaya56562