motioneye-project/motioneye #1843: Send motion notification to MULTIPLE webhook URLs

smithbill17, 2020-07-14:
I have **two** Android devices running an app called MacroDroid which is able to receive 'webhook' notifications and perform some kind of action.  One Android device is always online and one is sometimes offline (ie. at night).  So i wanted to configure MotionEye to send a motion notification to **TWO** webhook URLs so that both Android devices can be notified of camera motion.

I thought I had read somewhere that two webhook URLs can be entered into the appropriate field in the MotionEye UI separated using a ';' (semi-colon) character.  I tried this and found that I lost ALL of my configured cameras and could not undo the damage I had done   (I eventually found that I could modify  _/etc/motioneye/camera-1.conf_  on the server and delete the webhook URL i had entered and restart the server in order to restore my cameras).

Q. Is there a way to have MotionEye send a motion notification to MULTIPLE webhook URLs?

zagrim, 2020-07-14:
One way to call several URLs is to create a script accessible to MotionEye where you use e.g. `curl` or `wget` to call those webhook URLs, and put the path to the script to "Run A Command" option.

Thanks for mentioning potential field validation flaw, if MotionEye indeed fails to interpret the configuration when a webhook contains a semicolon. I'll try to remember to take a look at fixing that.

smithbill17, 2020-07-17:
I've got a bit of python code (below) which sends two separate webhook calls and can be invoked using the MotionEyes 'run command' option.

```
import requests
import time
t = time.localtime()
current_time = time.strftime("%H:%M:%S", t)
requests.post('https://trigger.macrodroid.com/<UUID1>/<ID1>?time=%s' % current_time)
requests.post('https://trigger.macrodroid.com/<UUID1>/<ID2>?timeporch=%s' % current_time)
```

So you can close this issue, or leave it open as a hook for fixing the webhook validation issue which caused all my cameras to disappear from the UI. Whichever is best for you.

Thanks.

zagrim, 2020-07-30:
What you were reading was the tooltip for notification commands, where it is specifically allowed to use a semicolon to separate multiple commands. In the web hook URL, however, a semicolon in the URL breaks the parsing of the config as you've seen. I created a PR which adds some validation to the web hook URL field to prevent this hitting anyone in the future.

zagrim, 2020-07-30:
I have no rights to close issues here, please do that since there's no use of keeping this open any longer.

closed 2020-07-30 by smithbill17