motioneye-project/motioneye #1844: Snapshot URL - 401: Unauthorized

allmoney, 2020-07-14:
I have problem in motioneye in Home Assistant.
"Snapshot URL" only works in same session.
If I open Snapshot URL in incognito chrome mode - I see error "401: Unauthorized".
Why? How fix?

[image]

GrinchUA, 2021-03-23:
I also think that this resource should be without authorization

starbasessd, 2021-03-23:
With a normal current install of motionEye or motionEyeOS, and you click on Snapshot URL, it gives you a link with the embedded username and hashed password to allow access:
http://[ip_address]/picture/1/current/?_username=admin_signature=d1231995ea72e4bb200a718b047a7c9fab677c06
which works in all browsers and works across reboots. Have you contacted the HomeAssistant support group?

starbasessd, 2021-03-23:
Works in Private or incognito modes also.

sargus123, 2021-04-06:
Links are provided differently than above, and unfortunately it does not work in a different browser and on the phone.

401

[image]
and : 

[image]

starbasessd, 2021-04-06:
Then contact the HA Community, and FRENCK, as he stated previously that his wrapper doesn't change anything, but it obviously does.

midnight1979, 2021-07-15:
use cookies in http request and it is woking!

closed 2021-07-15 by starbasessd