motioneye-project/motioneye #1845: Get access to Action buttons from external application

stefor9393, 2020-07-15:
Hi.
I have just discovered and install motioneyeOS on RPi 3B.
I have configured on my camera1 the preset1 action button to activate a relay to open / close my garage door remotely.
It works perfectly, but I would like to be able to call this action button directly from my domotic gateway.

I have tryed the following url
http://192.168.0.156/action/1/preset1/

but it answers "method not allowed".
I am pretty sure that this is an authentication issue, but I have no idea how to pass it.
Any Idea ?

davebdb, 2020-07-16:
You need to enable MOTION to accept non-local (127.0.0.1) connections:

Refer to the following:
Action buttons documentation here:
https://github.com/ccrisan/motioneye/wiki/Action-Buttons

Configuration file entry for "motion control"
https://github.com/ccrisan/motioneye/wiki/Configuration-File#motion_control_localhost....
```
motion_control_localhost
Set this to false to restrict the motion daemon's HTTP control server to listen on all interfaces. Defaults to true.
```
Hope that helps.  You can even set it up with ssh commands from your phone as well if you configure the hooks all correctly.

stefor9393, 2020-07-17:
Thanks for your reply davedb 👍 
I had the motion_control_localhost set to false in /data/etc/motion.conf but not in /etc/motioneye.conf.
This option was missing in motioneye.conf file, so I've add it at the end of the file, rebooted but still the "method not allowed" answer . And the action button still works fine on the web ui.

To be sure that it is not a remote restriction, I've tryied directly on the Rpi with SSH : 
curl http://127.0.0.1/action/1/preset1/
but I've got the same "message not allowed" message. Is the Url correct ?

stefor9393, 2020-07-17:
I had a look in debug mode in the browser and find this when I click on the action button 

http://192.168.0.156/action/1/preset1/?_=1594831863711&_username=admin&_signature=xxxxxxxxxxx
Both first and signature parameters are changing every time I click on the button. I believe this is for security purpose to pass credentials to the server.

stefor9393, 2020-07-19:
Nobody has any idea ? :(

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #972, which tracks an API for calling motionEye from other systems. Until then, the button is only a script, `/etc/motioneye/preset1_1` for preset1 on camera 1, so the gateway can run that same script directly, e.g. over SSH.

closed 2026-11-10 by JamBalaya56562