motioneye-project/motioneye #1851: Ubuntu 20.04 LTS

chrivosie, 2020-07-19:
Hello, 
i have installed motioneye on ubuntu 20.04 LTS and i can´t get the port 8765 to administration motioneye.

davebdb, 2020-07-19:
What error message are you getting when you try to web to it?  Can you send the output of 'netstat -atn' on the system it is running on?  Want to make sure that TCP/8765 is actually "LISTEN"ing before trying to troubleshoot further.

jptang2, 2020-07-26:
I have same problem. The TCP/8765 does not show up in 'netstat -atn'.

wombley, 2020-07-28:
Had this problem on Ubuntu 20.04 on a Raspberry Pi - apparently motioneye [hasn't been updated for Python 3](https://github.com/ccrisan/motioneye/issues/1640), and Ubuntu no longer includes Python 2 by default - and doesn't include pip for Python 2 at all. To fix:

`sudo pip3 uninstall motioneye`
`sudo apt install python2 python2-dev`
`curl https://bootstrap.pypa.io/get-pip.py --output get-pip.py `
`sudo python2 get-pip.py`
`sudo pip2 install motioneye`
`sudo systemctl start motioneye`

If this still doesn't work look in the logs immediatly after starting:
`sudo systemctl start motioneye`
`sudo tail -n 20 /var/log/syslog`

chrivosie, 2020-07-28:
Thx it works

bensekh, 2020-08-20:
I have installed MotionEye 0.4.21 successfully on Ubuntu 20.04 but the installation was a little different. At first I got error when adding PPA Repository for ffmpeg so I decided to install from Ubuntu Repo:
```
sudo apt-get install motion ffmpeg v4l-utils
```

Then added a universe repo:
```
sudo add-apt-repository universe
sudo apt update
```

Installed Python2
```
sudo apt install python2
curl https://bootstrap.pypa.io/get-pip.py --output get-pip.py
sudo python2 get-pip.py
sudo apt-get install python-dev-is-python2 curl libssl-dev libcurl4-openssl-dev libjpeg-dev
```
And make sure pip is using Python2
```
pip --version
```
Before start installing MotionEye, I installed the requirements with apt:
```
sudo apt install python-pillow
sudo apt install python-pycurl
```
Then, finally install MotionEye
```
sudo pip install motioneye
```
And do the rest according to the official instruction.

closed 2022-03-19 by MichaIng