motioneye-project/motioneye #1855: Can't stream by IP address.  Only by localhost

msmithGH, 2020-07-20:
Hi, 

Just installed MotionEye on Ubuntu 18.04 and love it so far.  Have one issue though.  I can access
http://localhost:8765 and http://localhost:808X
however going by local ip does not work.  e.g. http://192.168.86.10:8765 and http://192.168.86.10:808X does not work (not from other machines on the network and not even from the server).  I get "This site can't be reached.  192.168.86.10 refused to connect".  I also installed Deepstack on the same machine and I can freely access http://192.168.86.10:5000 from any machine on the network and on the server itself.

I opened motion.conf and stream_localhost to off and webcontrol_localhost to off and restarted motioneye, but it didn't make a difference.

How would I get this to work?

Thanks

msmithGH, 2020-07-20:
Well, I don't know what the real solution is, but I got it working by stopping the motioneye service and then installing the docker version.  It picked up all my cameras and settings so I didn't need to reconfigure everything, so it all just worked.

Thanks!

zagrim, 2020-07-21:
I wonder if there's a difference in the default value of `listen` in `/etc/motioneye/motioneye.conf`... Anyway, setting `listen 0.0.0.0` will bind MotionEye to all network interfaces, alternatively you might use `listen 192.168.86.10` if you want to be very specific.

JamBalaya56562, 2026-11-07:
Closing, as it works for you with the Docker image. For anyone with the same problem: `listen` in `motioneye.conf` sets the address of the web UI, and its default is `0.0.0.0`, all interfaces. A camera's streaming port only answers on other addresses with Video Streaming turned on in that camera's settings; otherwise it is bound to localhost.

closed 2026-11-07 by JamBalaya56562