motioneye-project/motioneye #1856: "Run a command" on motion detection not working (installed motioneye on Ubuntu)

t3rmin4l, 2020-07-22:
"Run a command" is not called so I have crontabs for rclone commands to sync with certain services pictures on motion detection. Not sure what's happening. Where do I begin? Do I have to give maybe special permissions to my script? Thanks a lot.

ancris00, 2020-08-07:
I have the same problem

JamBalaya56562, 2026-11-10:
Closing, as this got no details on the command and what `motioneye.log` shows at debug level when motion is detected. The command runs as the user motionEye and Motion run as, without a login shell, so it has to be executable by that user and use full paths. Please open a new issue with that log if a command still does not run with a current release.

closed 2026-11-10 by JamBalaya56562