motioneye-project/motioneye #1857: MotionEyeOS 4.2.2 not recording pictures or video

davidducic, 2020-07-23:
Hi everyone,

I've been using MotionEyeOS for about a year and it's worked perfectly until now. I used the setup from this video (https://www.youtube.com/watch?v=Qe-xs7Aqzt8), using a Raspberry Pi 4 with a Logitech webcam. I had to change my WiFi password yesterday, and when I did I decided to install the latest Motion Version and essentially do a clean install. Unfortunately, my ability to capture pictures and videos doesn't seem to be working now. I can take a snapshot and it will save on the MicroSD card installed in the Raspberry Pi, but that's it. I tried changing .mp4 to .avi but no joy with that, either.

Below are my settings - thanks in advance for your help:

**Video Device**

- Automatic Brightness: OFF
- Backlight Compensation: ON
- Brightness: 128
- Contrast: 32
- Exposure Absolute: 166
- Exposure Auto: 3
- Focus Absolute: 100
- Gain: 64
- Pan Absolute: 0
- Power Line Frequency: 2
- Saturation: 32
- Sharpness: 100
- Tilt Absolute: 0
- White Balance Temperature: 4000
- Zoom Absolute: 1
- Video Resolution: 1920x1080
- Frame Rate: 30

**File Storage**

- Storage Device: Custom Path
- Root Directory: /data/output/Camera1
- Upload Media Files: OFF
- Call A Web Hook: OFF
- Run A Command: OFF

**Video Streaming**

- Streaming Frame Rate: 30
- Streaming Quality: 100
- Streaming Image Resizing: OFF
- Streaming Port: 8081
- Authentication Mode: DISABLED
- Motion Optimization: OFF

**Still Images**

- Image Quality: 100%
- Capture Mode: Motion Triggered
- Preserve Pictures: Forever
- Enable Manual Snapshots: ON

**Movies**

- Movie Passthrough: OFF
- Movie Format: MPEG4 (.avi)
- Movie Quality: 100%
- Recording Mode: Motion Triggered
- Maximum Movie Length: 0
- Preserve Movies: Forever

**Motion Detection**

- Frame Change Threshold: 1%
- Maximum Change Threshold: 0 pixels
- Auto Threshold Tuning: OFF
- Auto Noise Detection: ON
- Light Switch Detection: 0%
- Despeckle Filter: ON
- Motion Gap: 30
- Captured Before: 30
- Captured After: 30
- Minimum Motion Frames: 20
- Mask: OFF
- Show Frame Changes: ON
- Create Debug Media Files: OFF

zagrim, 2020-07-23:
Any errors in the logs? Do you see enough changed pixels to match the 1% threshold? What if you add `emulate_motion on` in "Extra Motion Options", does that trigger video/picture capture?

nickmartin1ee7, 2020-09-16:
I just upgraded to this version, and I'm experiencing the same issue. My recording is set to continuous recording, and it rarely outputs any video files at all. This is for two separate devices

icurnow, 2020-11-08:
Same issue here after upgrade. I can click to take a pic this get correctly saved and sent to google. Motion shown by red frame results in nothing though....  Used to work fine.

hipitihop, 2021-04-07:
I have 2 cameras recording as expected, however a third, never records regardless of my motion settings. Not sure this is related.

starbasessd, 2021-04-07:
@hipitihop you don't give much info. 
Are your 3 cameras USB? If so, there's an inherent limit in Linux to 2 USB cameras per USB controller. Are they IP? WiFi or Ethernet connected? CSI Cable? Mix? What mix?
What SBC or PC are you using? Pi, (1,2,3B,3B+, 4B -1GB, -2GB, -4GB, -8GB, Zero, ZeroW) or one of a dozen clones, or PC?
Which OS are you running it on? 
Which install instruction set?
This is why it's a bad thing(TM) to hijack an existing issue. Better to mention the existing issue in a new one.

starbasessd, 2021-04-07:
@nickmartin1ee7 @icurnow Ditto to you, and at least answer @zagrim 's questions...
I use motion 4.2.2 for my production system (i5, Ubuntu 20.04, etc) and no issues with 5 cameras (1 USB Webcam, 3 d-Link DCS-5020-Ls, and a Foscam 9900 RTSP mode).

blgo, 2021-11-25:
I also had this issue on v0.42.1 and fixed it by changing the movie format to Matroska video .mkv

starbasessd, 2021-11-25:
Motion, not motionEye. Motion version should now be 4.3.2 (4.4.0 has issues) motionEye should be 0.42.1.

starbasessd, 2021-11-25:
Be careful if upgrading to Bullseye, also. They took out PiCamera, the raspivid and raspistill apps, and converted a lot of things to libcamera, which breaks things like motion. They also removed python camera too, breaking even more things.

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.42.1 and Motion 4.2.2, and the reports here differ: one user got movies again by changing the movie format to Matroska (.mkv), pointing to the encoder Motion used for MP4 on that system. zagrim's `emulate_motion on` test tells whether it is the detection or the recording. Please open a new issue with `motion.log` if a camera still records nothing with a current motionEye and Motion.

closed 2026-11-08 by JamBalaya56562