motioneye-project/motioneye #1858: Install On Raspbian failing at step 5

arturrocha, 2020-07-24:
```root@rp:~#  cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf```
```cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory```

I cant find /usr/local/share/motioneye at my raspberry pi.
Linux rp 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux

zagrim, 2020-07-25:
Searching issues with the error suggest that you might have not run `sudo pip install motioneye` with `sudo` as it says in the instructions, in that case the stuff would probably go to `.local/share/motioneye/extra` in your home directory instead of `/usr/local/share/motioneye/extra` (and possibly break some other things, although I'm not sure of that).

arturrocha, 2020-07-25:
Not really, i ran as sudo. 
It was a try on the new Pi OS. I switched to 2019-09-26-raspbian-buster and the install works fine.

zagrim, 2020-07-29:
Well, with `find / -name motioneye.conf.sample` you'll be able to find it, but my bet for its location is `.local/share/motioneye/extra` for whatever reason.

UrbanSide, 2020-08-15:
+1 i have this error on the opi lite

cxbrooks, 2020-08-17:
I had similar problems where after `pip install motioneye` /usr/local/share/motioneye did not exist.

Oddly, it seemed like each time I ran `pip install motioneye`, it would get a little further.  I was logged in as root via `sudo -i` under buster.

As per [the instructions](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian), I did

```
apt-get install python-pillow
pip install motioneye
```
and /usr/local/share/motioneye was present.

Below is the complete run, showing how `pip install motioneye` did not create /usr/local/share/motioneye and how it was later created:

```
root@raspberrypi:~# pip install motioneye
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
  Using cached https://files.pythonhosted.org/packages/c5/ff/139b4b02ed089b2b50eee65d1059ae0d290\
64ab5f4ad429e45c702396697/motioneye-0.42.1.tar.gz
Collecting jinja2 (from motioneye)
  Using cached https://files.pythonhosted.org/packages/30/9e/f663a2aa66a09d838042ae1a2c5659828bb\
9b41ea3a6efa20a20fd92b121/Jinja2-2.11.2-py2.py3-none-any.whl
Collecting pillow (from motioneye)
  Downloading https://files.pythonhosted.org/packages/b3/d0/a20d8440b71adfbf133452d4f6e0fe80de2d\
f7c2578c9b498fb812083383/Pillow-6.2.2.tar.gz (37.8MB)
    99% |████████████████████████████████| 37.8MB 826kB/s eta 0:00:01Killed
root@raspberrypi:~# ls /usr/local/share/
ca-certificates  emacs  fonts  man
root@raspberrypi:~# apt-get install python-pillow
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python-pil' instead of 'python-pillow'
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'apt autoremove' to remove it.
The following additional packages will be installed:
  libimagequant0 libwebpdemux2 python-olefile
Suggested packages:
  python-pil-doc python-pil-dbg
The following NEW packages will be installed:
  libimagequant0 libwebpdemux2 python-olefile python-pil
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 514 kB of archives.
After this operation, 1,672 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://raspbian.freemirror.org/raspbian buster/main armhf libimagequant0 armhf 2.12.2-1.1 \
[27.2 kB]
Get:2 http://raspbian.freemirror.org/raspbian buster/main armhf libwebpdemux2 armhf 0.6.1-2 [86.\
7 kB]
Get:3 http://raspbian.freemirror.org/raspbian buster/main armhf python-olefile all 0.46-1 [35.9 \
kB]
Get:4 http://raspbian.freemirror.org/raspbian buster/main armhf python-pil armhf 5.4.1-2+deb10u2\
 [364 kB]
Fetched 514 kB in 3s (155 kB/s)
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a \
controlling terminal.)
debconf: falling back to frontend: Readline
electing previously unselected package libimagequant0:armhf.
(Reading database ... 47882 files and directories currently installed.)
Preparing to unpack .../libimagequant0_2.12.2-1.1_armhf.deb ...
Unpacking libimagequant0:armhf (2.12.2-1.1) ...
Selecting previously unselected package libwebpdemux2:armhf.
Preparing to unpack .../libwebpdemux2_0.6.1-2_armhf.deb ...
Unpacking libwebpdemux2:armhf (0.6.1-2) ...
Selecting previously unselected package python-olefile.
Preparing to unpack .../python-olefile_0.46-1_all.deb ...
Unpacking python-olefile (0.46-1) ...
[... 36 lines cut ...]
  Downloading https://files.pythonhosted.org/packages/c5/10/369f50bcd4621b263927b0a1519987a04383\
d4a98fb10438042ad410cf88/singledispatch-3.4.0.3-py2.py3-none-any.whl
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from singledispatch->tor\
nado<6,>=3.1->motioneye) (1.12.0)
Building wheels for collected packages: motioneye, pycurl, tornado, MarkupSafe
  Running setup.py bdist_wheel for motioneye ... done
  Stored in directory: /root/.cache/pip/wheels/b1/b5/20/266f8871b2b1f598c96a17e5545a16f2c2198956\
890f17fb5e
  Running setup.py bdist_wheel for pycurl ... done
  Stored in directory: /root/.cache/pip/wheels/98/a3/19/9439c79b88a742f508af0c4e410ae3f685db2e13\
d3374e798e
  Running setup.py bdist_wheel for tornado ... done
  Stored in directory: /root/.cache/pip/wheels/6d/e1/ce/f4ee2fa420cc6b940123c64992b81047816d0a9f\
ad6b879325
  Running setup.py bdist_wheel for MarkupSafe ... done
  Stored in directory: /root/.cache/pip/wheels/f2/aa/04/0edf07a1b8a5f5f1aed7580fffb69ce8972edc16\
a505916a77
Successfully built motioneye pycurl tornado MarkupSafe
Installing collected packages: MarkupSafe, jinja2, pycurl, backports-abc, futures, singledispatc\
h, tornado, motioneye
Successfully installed MarkupSafe-1.1.1 backports-abc-0.5 futures-3.3.0 jinja2-2.11.2 motioneye-\
0.42.1 pycurl-7.43.0.5 singledispatch-3.4.0.3 tornado-5.1.1
root@raspberrypi:~# ls /usr/local/share/
ca-certificates  doc  emacs  fonts  man  motioneye
```

mgjsmith94, 2020-08-18:
On my Pi 4 I found them both (all) located in a  Hidden Folder 
  root    /home/pi/.local/share/motioneye/extra/
Open the File Explorer to -home/pi-   then type /.local  & press Enter to check if they are there  

If they are adjust the CP command to reflect this location .

MGJS

Docteur-Who, 2021-01-14:
Hi,
after a fresh install on a Pi Zero W and Raspian, I confirm to have the same problem of location of the file motioneye.conf.sample
The correct location is 
./tmp/pip-install-ZYmKDl/motioneye/extra/motioneye.conf.sample
Same thing with motioneye.service

Hope to be helpfull

starbasessd, 2021-01-14:
Using the wrong pip. Please try the Ubuntu 20.04 or newer instructions.
I haven't updated all the instructions yet, due to Upstream sources not
supporting pip2 and they are not giving downstream warnings.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-05:
Confirmed instructions 
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
working 3/3/2021.
Closing

closed 2021-03-05 by starbasessd