motioneye-project/motioneye #1859: Videos too heavy

Saku241, 2020-07-24:
Hello 
I run MotionEye 0.42.1 with Motion 4.3.1 on Raspbian 10

I record video on motion with H264/OMX on quality 100% on resolution 1280x720@10 fps and video are quite heavy

Like each second of video its equal to an average of 4MB 

I quietly have multiples videos of 60+MB

Do you know a better solution to fix this ? Another codec or anything ?

Thanks

zagrim, 2020-07-25:
Don't record with quality set quite as high as 100%? :wink: I mean, a slight drop in quality might mean a significant drop in size.

I don't think there are any other codecs that would have hardware support on Raspberry Pi, so anything else will probably be too heavy for the CPU.

Saku241, 2020-07-25:
> Don't record with quality set quite as high as 100%? 😉 I mean, a slight drop in quality might mean a significant drop in size.
> 
> I don't think there are any other codecs that would have hardware support on Raspberry Pi, so anything else will probably be too heavy for the CPU.

If I put a value under 100%, it starts to drop frames due to compression and a more cpu usage

zagrim, 2020-07-25:
I haven't actually ever thought that 100% might somehow by-pass compression. Live and learn. Then again, my setup uses `movie_passthrough` option to completely avoid needing to encode on the Pi. If you don't need overlays (text or privacy masking by MotionEye) and you have [a suitable camera](https://motion-project.github.io/motion_config.html#movie_passthrough) it might also work for you, though.

Saku241, 2020-07-25:
> I haven't actually ever thought that 100% might somehow by-pass compression. Live and learn. Then again, my setup uses `movie_passthrough` option to completely avoid needing to encode on the Pi. If [...]

Also I don’t understand because when I switch on movie passthrought on MotionEye I still get videos with text overlay and such things. 

But I’ll try again

zagrim, 2020-07-25:
The Motion configuration document I linked should give all the details to be able to successfully use movie passthrough, I think. Did you also notice that the camera "must be specified using the `netcam_url` parameter and the `v4l2` prefix"?

Saku241, 2020-07-25:
> The Motion configuration document I linked should give all the details to be able to successfully use movie passthrough, I think. Did you also notice that the camera "must be specified using the `netcam_url` parameter and the `v4l2` prefix"?

```
URL to use if you are using a network camera, size will be autodetected. Available prefixes to the URL:
```

How should I use it ? Since I don’t use a network camera it’s directly using USB on my Pi

Saku241, 2020-07-25:
Ok I found a solution at this problem 

• I need to remove the movie passthought because It gives me too bigger files but the frame rate is amazing. (Approx. 5MB/s)

• I put again H264/OMX but I reduced the movie quality to 70%. The frame rate is worse than 100% (like approx. 8 fps) BUT the size is about 0.7MB/s !

So I guess I need to sacrifice some FPS to a much smaller video size, that sounds logic but still...

JamBalaya56562, 2026-11-07:
Closing as solved, as you found a balance: H.264/OMX with the movie quality at 70% gives about 0.7 MB/s, at the cost of some frame rate.

closed 2026-11-07 by JamBalaya56562