motioneye-project/motioneye #1861: MotionEye not creating log file

Pi-DD, 2020-07-25:
This is, most likely, user error!! But I’ve set up everything carefully as far as I can tell but no motioneye.log file appears to be created.

Guidance appreciated!

zagrim, 2020-07-26:
Does the system you've installed MotionEye use `systemd`? If so see https://github.com/ccrisan/motioneye/wiki/Troubleshooting#getting-the-log-file on how to get the logs out.

Pi-DD, 2020-07-26:
Thank you! Yes, that did indeed create a log file in my home directory.  

I had expected, from reading the wiki, that such a file would automatically and continually be created in one of the directories listed in the configuration file description for “log-path”, but could not find it in any of them,

zagrim, 2020-07-27:
Yes, well, it is quite the same with all software run as a service in a Linux distribution: if `systemd` (and thus `systemd-journald`) are being used, all logging will go to system journal instead of text files.

I guess there still could be a note in the Configuration wiki page mentioning this, too. 
@ccrisan, how about adding something like "When running on a system with `systemd`, logs will go to the system journal and are available via `journalctl`." to https://github.com/ccrisan/motioneye/wiki/Configuration-File under "log_path" (and maybe also to `motioneye.conf.sample`)?

Pi-DD, 2020-07-27:
That would be helpful. As written, for those of us on a “learning curve” (!) with Linux the Wiki gives a different impression.

closed 2020-07-27 by Pi-DD