motioneye-project/motioneye #1866: Can not connect to Motioneye web interface when installed on Raspberry Pi 4 (2GB), Raspian (buster).

HunterAhlquist, 2020-07-28:
Hi, I followed the instructions exactly on [this webpage](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian).
I am using a Raspberry Pi 4B (2GB RAM)

This is what I get when I enter `journalctl | grep -E "meyectl|motion"`:

`Jul 27 19:02:37 raspberrypi motion[342]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Jul 27 19:15:38 raspberrypi sudo[898]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
Jul 27 19:15:51 raspberrypi sudo[967]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/mkdir -p /var/lib/motioneye
Jul 27 19:16:25 raspberrypi sudo[1108]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
Jul 27 19:17:09 raspberrypi sudo[1323]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/systemctl enable motioneye
Jul 27 19:17:31 raspberrypi sudo[1441]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/systemctl start motioneye
Jul 27 19:17:31 raspberrypi systemd[1]: Started motionEye Server.
Jul 27 19:17:31 raspberrypi meyectl[1448]:     INFO: hello! this is motionEye server 0.42.1
Jul 27 19:17:33 raspberrypi meyectl[1448]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Jul 27 19:17:33 raspberrypi meyectl[1448]:     INFO: cleanup started
Jul 27 19:17:33 raspberrypi meyectl[1448]:     INFO: wsswitch started
Jul 27 19:17:33 raspberrypi meyectl[1448]:     INFO: tasks started
Jul 27 19:17:33 raspberrypi meyectl[1448]:     INFO: mjpg client garbage collector started
Jul 27 19:17:33 raspberrypi meyectl[1448]:     INFO: server started`

It seems like it's functioning, but when I try to access the web interface I am unable to connect on all devices and browsers I try. I can only get the software to work when i use MotionEyeOS. I made sure that I was connecting to the HTTP port and not the HTTPS port.
Is this an incompatibility with Raspbian buster?

zagrim, 2020-07-28:
I'm thinking that a configuration issue is the cause. How are you trying to connect to the web interface, locally on the same computer where you installed MotionEye, or from another box? What output do you get if you run this command on the Pi: `sudo netstat -tnop | grep 8765` ?

jmichault, 2020-08-01:
it looks like you had a problem in step 5, you should do it again:
```
sudo mkdir -p /etc/motioneye
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
```

Nickybroke, 2020-08-13:
Same here:
 motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2020-08-13 15:05:27 CEST; 2min 27s ago
  Process: 15645 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=killed, signal=SEGV)
 Main PID: 15645 (code=killed, signal=SEGV)

Aug 13 15:05:27 raspberrypi systemd[1]: motioneye.service: Service RestartSec=100ms expired, scheduling restart.
Aug 13 15:05:27 raspberrypi systemd[1]: motioneye.service: Scheduled restart job, restart counter is at 5.
Aug 13 15:05:27 raspberrypi systemd[1]: Stopped motionEye Server.
Aug 13 15:05:27 raspberrypi systemd[1]: motioneye.service: Start request repeated too quickly.
Aug 13 15:05:27 raspberrypi systemd[1]: motioneye.service: Failed with result 'signal'.
Aug 13 15:05:27 raspberrypi systemd[1]: Failed to start motionEye Server.

closed 2021-03-05 by starbasessd