motioneye-project/motioneye #1867: Motion Notification End Command runs twice for no reason on config save

DaniW42, 2020-07-29:
It looks like an "End Command" which is configured to run after a motion is detected runs twice as soon as i save any configuration changes inside the web interface. My "End Command" script also writes to an log file and as soon as I hit the Apply button i get overlapping entries in the log file.

    Do 30. Jul 00:43:51 CEST 2020 - MOTION DETECTED.
    Do 30. Jul 00:43:51 CEST 2020 - MOTION DETECTED.
    Do 30. Jul 00:43:53 CEST 2020 - OP6T available, end script.
    Do 30. Jul 00:43:53 CEST 2020 - OP6T available, end script.

which normally should look like 

    Do 30. Jul 00:43:51 CEST 2020 - MOTION DETECTED.
    Do 30. Jul 00:43:51 CEST 2020 - OP6T available, end script.
    Do 30. Jul 00:43:53 CEST 2020 - MOTION DETECTED.
    Do 30. Jul 00:43:53 CEST 2020 - OP6T available, end script.

Of course, in the end it should not run at all as no motion is visible and even no media-file is being created. But the log clearly shows that it runs the command simultaneously.

starbasessd, 2021-02-04:
> Got the same issue using this version :
>
> OS_NAME="motionEyeOS"
> OS_SHORT_NAME="motioneyeos"
> OS_PREFIX="meye"
> OS_VERSION="20200606"

Have you tried dev20201026? There are some weird glitches with 20200606
Whqat qre your versions of motion, motioneye, and what is your base OS?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DBarthe, 2021-02-04:
> Have you tried dev20201026? There are some weird glitches with 20200606 Whqat qre your versions of motion, motioneye, and what is your base OS?

I'm using the pre-installed OS from motioneyeos, the latest stable version downloaded from the github repository. I'm not able to upgrade right now. Just wanted to confirm this issue. :)

starbasessd, 2021-03-05:
Let me know when you can upgrade to dev20201026 or downgrade to 20190911 (if not using a Pi4)

JamBalaya56562, 2026-11-08:
Closing this, as it does not seem to happen any more. When the settings are saved, motionEye stops Motion and waits for it to exit before starting it again; with the current code and Motion 4.7, saving the settings of a camera three times in a row still left a single Motion process, with each event command once in its config. The duplicated lines look like two Motion processes running at once, possibly one of the glitches of motionEyeOS 20200606 mentioned above. If you still see a command run twice on motionEye 0.43.1 or newer, please open a new issue with the motionEye and Motion versions.

closed 2026-11-08 by JamBalaya56562