motioneye-project/motioneye #1869: Installation ubuntu 20 in Proxmox

SmartyML, 2020-07-30:
Hallo,

ich versuche einen MotionEye Server auf Ubuntu 20 aufzusetzen.
Basis ist ein Proxmox LX Container, auf gleichen läuft bereits erfolgrei Grafana und einer mit iobroker...

Nachfolgendes führe ich nach Erstinstall aus:

`
sudo apt-get update && sudo apt-get upgrade

sudo apt install python2
`

Hier verlangt die Anleitung später pip oder pip2 beide führen zu folgendem Fehler:

`
...:apt install python-pip
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python-pip
`

Es klappt dann weiter:

`
apt install python3-pip^C
`

Dann geht es weiter nach der Anleitung:
https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu

`
apt-get install motion ffmpeg v4l-utils
apt-get install python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev
`

Dann, da pip nicht installiert:

`
pip3 install motioneye

mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf

mkdir -p /var/lib/motioneye

cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
`

Anleitung ist ende und ich habe keinen Zugriff auf Motioneye.

`
root@MotionEye:~# ps -e
  PID TTY          TIME CMD
    1 ?        00:00:00 systemd
   49 ?        00:00:00 systemd-journal
   75 ?        00:00:00 systemd-network
   96 ?        00:00:00 accounts-daemon
   97 ?        00:00:00 cron
   98 ?        00:00:00 dbus-daemon
  101 ?        00:00:00 networkd-dispat
  102 ?        00:00:00 rsyslogd
  103 ?        00:00:00 systemd-logind
  104 ?        00:00:00 systemd-resolve
  123 console  00:00:00 agetty
  124 pts/0    00:00:00 login
  125 pts/1    00:00:00 agetty
  132 ?        00:00:00 sshd
  277 ?        00:00:00 master
  278 ?        00:00:00 pickup
  279 ?        00:00:00 qmgr
  306 pts/0    00:00:00 bash
 8454 pts/0    00:00:00 ps
`

Nun weiß ich nicht weiter. Ich kann keine Fehlmeldungen entdecken.
Könnt Ihr helfen?

zagrim, 2020-07-30:
Please try the solution for pip2 not available in Ubuntu 20 given here: #1851.

SmartyML, 2020-08-01:
Python2 und pip could be installed.
During the installation von MotionEye there ist still an error.

`
root@MotionEye:~#  pip install motioneye
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting motioneye
  Downloading motioneye-0.42.1.tar.gz (758 kB)
     |################################| 758 kB 1.3 MB/s 
Collecting tornado<6,>=3.1
  Downloading tornado-5.1.1.tar.gz (516 kB)
     |################################| 516 kB 3.1 MB/s 
Collecting jinja2
  Downloading Jinja2-2.11.2-py2.py3-none-any.whl (125 kB)
     |################################| 125 kB 3.1 MB/s 
Collecting pillow
  Downloading Pillow-6.2.2-cp27-cp27mu-manylinux1_x86_64.whl (2.1 MB)
     |################################| 2.1 MB 3.6 MB/s 
Collecting pycurl
  Downloading pycurl-7.43.0.5.tar.gz (216 kB)
     |################################| 216 kB 3.7 MB/s 
Collecting futures
  Downloading futures-3.3.0-py2-none-any.whl (16 kB)
Collecting singledispatch
  Downloading singledispatch-3.4.0.3-py2.py3-none-any.whl (12 kB)
Collecting backports_abc>=0.4
  Downloading backports_abc-0.5-py2.py3-none-any.whl (5.2 kB)
Collecting MarkupSafe>=0.23
  Downloading MarkupSafe-1.1.1-cp27-cp27mu-manylinux1_x86_64.whl (24 kB)
Collecting six
  Downloading six-1.15.0-py2.py3-none-any.whl (10 kB)
Building wheels for collected packages: motioneye, tornado, pycurl
  Building wheel for motioneye (setup.py) ... done
  Created wheel for motioneye: filename=motioneye-0.42.1-py2-none-any.whl size=790692 sha256=736aa1eeb4c3da6a964211e7d8078a7836eaa3380bddca9a6d916de49bb17bce
  Stored in directory: /root/.cache/pip/wheels/24/45/97/6cdec7430d1871c0a66171972c604033182ab104be32351e89
  Building wheel for tornado (setup.py) ... done
  Created wheel for tornado: filename=tornado-5.1.1-cp27-cp27mu-linux_x86_64.whl size=449883 sha256=d5dcfb598747818900372512749871996d7b428ed410fe3ebca1a5eed1bac7fc
  Stored in directory: /root/.cache/pip/wheels/d8/83/af/e0dc6afbf3a2c51af8d6e3f9fbe790d0c581c2de05bc5d50f5
  Building wheel for pycurl (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-dqPv9s/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-dqPv9s/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-h4Enut
       cwd: /tmp/pip-install-dqPv9s/pycurl/
  Complete output (16 lines):
  Using curl-config (libcurl 7.68.0)
  Using SSL library: OpenSSL/LibreSSL/BoringSSL
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-2.7
  creating build/lib.linux-x86_64-2.7/curl
  copying python/curl/__init__.py -> build/lib.linux-x86_64-2.7/curl
  running build_ext
  building 'pycurl' extension
  creating build/temp.linux-x86_64-2.7
  creating build/temp.linux-x86_64-2.7/src
  x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-1x6jhf/python2.7-2.7.18~rc1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPYCURL_VERSION="7.43.0.5" -DHAVE_CURL_SSL=1 -DHAVE_CURL_OPENSSL=1 -DHAVE_CURL_SSL=1 -I/usr/include/python2.7 -c src/docstrings.c -o build/temp.linux-x86_64-2.7/src/docstrings.o
  unable to execute 'x86_64-linux-gnu-gcc': No such file or directory
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for pycurl
  Running setup.py clean for pycurl
Successfully built motioneye tornado
Failed to build pycurl
DEPRECATION: Could not build wheels for pycurl which do not use PEP 517. pip will fall back to legacy 'setup.py install' for these. pip 21.0 will remove support for this functionality. A possible replacement is to fix the wheel build issue reported above. You can find discussion regarding this at https://github.com/pypa/pip/issues/8368.
Installing collected packages: futures, six, singledispatch, backports-abc, tornado, MarkupSafe, jinja2, pillow, pycurl, motioneye
    Running setup.py install for pycurl ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-dqPv9s/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-dqPv9s/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-hkez4f/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python2.7/pycurl
         cwd: /tmp/pip-install-dqPv9s/pycurl/
    Complete output (16 lines):
    Using curl-config (libcurl 7.68.0)
    Using SSL library: OpenSSL/LibreSSL/BoringSSL
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-2.7
    creating build/lib.linux-x86_64-2.7/curl
    copying python/curl/__init__.py -> build/lib.linux-x86_64-2.7/curl
    running build_ext
[... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-dqPv9s/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-dqPv9s/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-hkez4f/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python2.7/pycurl Check the logs for full command output.
`

jmichault, 2020-08-01:
Hallo,
ist gcc installiert?

`sudo apt-get install gcc`

SmartyML, 2020-08-02:
Ja ist es.

jmichault, 2020-08-02:
Hallo,
«unable to execute 'x86_64-linux-gnu-gcc': No such file or directory» ist normalerweise ein Zeichen dafür, dass gcc nicht installiert ist. Versuchen Sie es neu zu installieren :
`sudo apt reinstall gcc`

closed 2021-03-05 by starbasessd