motioneye-project/motioneye #1871: Motioneye in docker no networkshare possible

holzfelix, 2020-07-31:
I try to install and run motioneye in docker. I have set it up with docker-compose 

[image]

`---
version: "3.5"
services:
  motioneye:
    image: ccrisan/motioneye:master-amd64  # Change to ccrisan/motioneye:master-armhf for ARM chips (Pi etc.)
    ports:
      - "8081:8081"
      - "8765:8765"
    volumes:
      - etc_motioneye:/etc/motioneye
      - var_lib_motioneye:/var/lib/motioneye

volumes:
  etc_motioneye:
  var_lib_motioneye:`

After this I imported my config from raspberrypi and now I can't select network devices as file storage.
[image]

How can I solve this?

Thanks.

kleini, 2020-07-31:
You should mount the network file storages on the Docker host and mount the host filesystem then into the container. Network mounts in containers are discouraged. Keep in mind, that a container is meant to run a single process, not a whole Linux operating system.

closed 2020-07-31 by kleini

holzfelix, 2020-07-31:
Thats correct didn't thought about it. Thanks